EdgeTX/edgetx #1204: Persistent Bluetooth - remember devices/autoconnect

IdefixRC, 2021-12-02:
Not sure if this is technically possible (store the BT device and autoconnect) but assume is should be doable?

**Is your feature request related to a problem? Please describe.**
I'm always frustrated when....I have to rebind my Bluetooth head tracker each time I restart the TX16 Transmitter. I'm wondering if there is a way for the radio to remeber the device and auto connect to it.

**Describe the solution you'd like**
Bind to a BT device and upon restart the radio checks whether the device can be found and either auto connects or provides a pop-up with the question: XX Found, Reconnect? Yes/No

dlktdr, 2022-01-04:
I will take a look at this. It used to work in OpenTX so should here too.. can't believe I didn't even notice. Constantly switching back and forth while testing and miss the basics.

I had an issue on the Headtracker project where even in OTX it wouldn't always reconnect but only with the non-para BT module(CC2540). Out of curiosity which module do you have PARA (CC2650) or Bluetooth Trainer(CC2540).

Either module it should be possible.

IdefixRC, 2022-01-04:
Thanks. It didn't work for me in OpenTX either.
I'm using it with an Arduino nano 33 ble sense (head tracker) but no idea
what BT chipset it has unfortunately.