EdgeTX/edgetx #1205: YAML: reading mixer from model.yml fails reading mixer-script definitions

wimalopaan, 2021-12-02:
If a model has the following mixers (see srcRaw!) with twi different mixer-script sources:

```
mixData:
-
weight: 100
destCh: 0
srcRaw: lua(lua(0,1)
carryTrim: 0
mixWarn: 0
mltpx: ADD
offset: 0
swtch: NONE
flightModes: 000000000
delayUp: 0
delayDown: 0
speedUp: 0
speedDown: 0
name: ""
-
weight: 100
destCh: 1
srcRaw: lua(lua(0,0)
carryTrim: 0
mixWarn: 0
mltpx: ADD
offset: 0
swtch: NONE
flightModes: 000000000
delayUp: 0
delayDown: 0
speedUp: 0
speedDown: 0
name: ""
```
these both (or: all) are always read on the next start of the radio as:

```
srcRaw: lua(lua(0,0)
```

Maybe this is an error according to the missing quoting here?

wimalopaan, 2021-12-02:
Maybe related to #1131 ???

raphaelcoeffic, 2021-12-03:
`lua(` should be there only once. This is probably a stupid mistake while generating the value.

eshifri, 2021-12-03:
https://github.com/EdgeTX/edgetx/blob/main/radio/src/storage/yaml/yaml_datastructs_funcs.cpp#L284

PR #1208 (2021-12-04): Fixed "lua(...,...)" source
Resolves #1205
commit 995e4146 (2021-12-04): Fixed "lua(...,...)" source (#1208)

Resolves #1205
closed 2021-12-04 by pfeerick