EdgeTX/edgetx #1214: NV14: Actual Nightlys have no internal RF module available

uPz3TT3r, 2021-12-05:
On NV14 (Flysky Nirvana) i have tested the latest nightly builds (01.12.21 and 05.12.21) and there is no internal RF module available.
"OFF" is the only option.
I tested it with my two Nirvanas, both had the problem.
After that i installed Release 2.5.0 and the internal RF module is available in 

**To Reproduce**
1. Install Nightly
2. Go to model setup
3. Change Internal RF Mode
4. OFF is only option

**Expected behavior**
Expected is to show the internal Flysky module. (Works with 2.5.0 Release)

**General information**
 - EdgeTX version (2021-12-05T01:09:08Z)
 - Hardware (Flysky NV14)

pfeerick, 2021-12-05:
I think you need to go into the radio setup -> hardware tab and set the internal module to flysky - for some reason yours has turned off. I'll have to check what is going on there - either the upgrade path from 2.5 is borked, or it's playing up because you've been on nightlies, which is fine if that is the case.

pfeerick, 2021-12-06:
I just upgraded from 2.5 to the current nightly, and FLYSKY was enabled by default. Can you confirm if you have been running nightlies prior to this not working? (meaning the configuration was most likely broken because you didn't update directly from 2.5 to the nightly that introduced the new option).

uPz3TT3r, 2021-12-06:
I havent known that there is another point where the Module could be turned off. On the hardware tab it was off and could be enabled, now it shows up in the model settings too. Thank you!

i updated from opentx directly to nightly 2021-12-01 and the internal modul was Not available in model tab. After that i updated to 2.5.0 and the module was available, then i updated to nightly 2021-12-05 and it was not available. I never changed it in Hardware tab because i dont know that this exists.

pfeerick, 2021-12-06:
> i updated from opentx directly to nightly 2021-12-01

Hm, that is probably something to do with it... may have to see if we force it to be FLYSKY for the NV14 since it is currently the only option (other radios have CRSF, XJT, etc as options). This is a new feature for 2.6.0.

raphaelcoeffic, 2021-12-10:
> > i updated from opentx directly to nightly 2021-12-01
> 
> Hm, that is probably something to do with it... may have to see if we force it to be FLYSKY for the NV14 since it is currently the only option (other radios have CRSF, XJT, etc as options). This is a new feature for 2.6.0.

Shouldn't your fix in 15a6b6f take care of this?

pfeerick, 2021-12-10:
Lol... yes, I forgot to link the issue as that is what triggered it... ty ;)

closed 2021-12-10 by pfeerick