EdgeTX/edgetx #1215: NV14: Actual Nightlys - widget fullscreen mode cant be closed

uPz3TT3r, 2021-12-05:
**Describe the bug**
On Flysky Nirvana (NV14) a fullscreen widget cant be closed.

**To Reproduce**
1. Click on a widget (in my case model or outputs widget)
2. select "Fullscreen"
3. The widget is shown in fullscreen mode, but no chance to leave this mode
4. Only a restart of NV14 could leave fullscreen mode

**Expected behavior**
Another click on the touch should leave Fullscreen mode?!

**General information**
 - EdgeTX version (nightly 2021-12-05T01:09:08Z, also tested with nightly 2021-12-01...)
 - Hardware (Flysky Nirvana, NV14)

eshifri, 2021-12-05:
This is a duplicate of #821.

Also please notice that long press on the left trim button closes the full screen widget.

pfeerick, 2021-12-05:
Duplicate of #821

closed 2021-12-05 by pfeerick