EdgeTX/edgetx #1216: Companion 2.5 autoOFF backlight issue on LED screen

andreapede, 2021-12-05:
In case of a new radio setting created in companion from the scratch, the "Backlight AutoOFF" value is set to 0 sec.
So pushing the setting as is in the radio, the radio will show a black screen.

Probably the default settings should be different from 0 or completely exclued for interested class of  radio models.

EdgeTX 2.5 Companion
Radiomaster T16S

pfeerick, 2021-12-06:
This was the default setting when I simply hit the 'new' option in Companion, and had a look at the default settings:
[image]

This means the OFF setting will always be applied, and since that was 20, it will only be dim, not off. But I do agree that it should not default to zero - more like 15 seconds.

elecpower, 2021-12-06:
Should mirror radio defaults

pfeerick, 2021-12-06:
True - it should, but it doesn't - on TX16S,  backlightMode and blOffBright are both correct, lightAutoOff  should be 10 seconds as far as the user is concerned. 

The actual value of lightAutoOff is `g_eeGeneral.lightAutoOff = 2;` (as set by `generalDefault()`, but naturally, as you _would_ do, it is then multiplied by 5 / divided by 5 later in the code, hence 10 seconds.  🤦 

```
    auto edit = new NumberEdit(window, grid.getFieldSlot(2, 1), 5, 600,
                               GET_DEFAULT(g_eeGeneral.lightAutoOff * 5),
                               SET_VALUE(g_eeGeneral.lightAutoOff, newValue / 5));
```

elecpower, 2021-12-06:
One of many bit minimisatios for those eeproms. Need to get rid of these especially with  YAML storage so what you see in the yaml file is what you see on radio and Companion.

PR #1228 (2021-12-09): Align Companion radio default settings
Fixes #1216

Summary of changes:
Align Companion radio settings defaults to firmware
closed 2021-12-09 by pfeerick