EdgeTX/edgetx #1217: Model Select link for Widgets

Carl-RC, 2021-12-06:
When selecting a model image for a Widget, if the link mechanism to see the list of images is lost, there’s no apparent, obvious way to recover the link?
What was list; a file or a variable?

**General information**
 - EdgeTX version (e.g. 2.4.0 or nightly and #commit number)
 - Hardware (e.g. Frsky X9D+ or RM TX12S)

pfeerick, 2021-12-06:
I'm not quite sure what you mean there. In the model setup, you can pick an image for the model from those contained in the `\IMAGES` folder, as long as the names are not too long (of the top of my head, it's something like 8-12 characters long + ext). That is saved into the model configuration as a setting. If there are no options, this means you have files with too long a name, the wrong extension/format or nothing in the `\IMAGES` folder (assuming there isn't a bug). 

Can you provide more information as to what is the actual issue? And on what firmware version and hardware?

Carl-RC, 2021-12-06:
It was working and for some reason I deleted whatever was in the text box left of the list arrow; it never worked after that and I can’t enter anything in the box.
It was working before I did that; I’m not sure what I deleted and it would seem that it shouldn’t have been editable by a user.

Carl-RC, 2021-12-06:
The version is the latest, I just downloaded it a couple days ago.

Carl-RC, 2021-12-06:
All the images have short names, and had been used in widgets before.

pfeerick, 2021-12-06:
Ok, so are we talking about radio or companion? As it sounds like you're talking about Companion, and the list represents the contents of the `IMAGES` subfolder of the SD Structure Path / or what you set it on radio if it is configured. If this is the case, you need to either have a copy of your SD card on your computer and `Settings -> Settings ... -> SD Structure path` pointing to it, or need to select it again in the model settings screen on the radio.

Carl-RC, 2021-12-06:
Yes, I should have said that; but, if I try to do it on the RadioMaster T16S, no luck there either? The images are in the folder, they worked at one time. I don’t know if a file of some type got deleted and where it would be?

pfeerick, 2021-12-06:
So do *any* images get listed when you try to set the model image on the radio? As there is no file to be deleted - it should be parsing the IMAGES directory directly for compatible file names and lengths.

Carl-RC, 2021-12-06:
No,  none show up.

Carl-RC, 2021-12-06:
They use too!

Carl-RC, 2021-12-06:
It’s like the radio lost the path to the image folder?

pfeerick, 2021-12-06:
Sounds more like the card is corrupt or the files deleted. Can you see the images if you use the RADIO/SYS SD card browser?

Carl-RC, 2021-12-06:
If I connect the radio to my PC thru the USB port, I can bring all the images up. They are still good, same images that worked before.

Carl-RC, 2021-12-06:
I used the windows file explorer to open them; it shouldn’t make any difference?

Carl-RC, 2021-12-06:
I used the radio browser also, the files are in the image folder.

Carl-RC, 2021-12-06:
When the firmware is flashed to the radio, are there any default variable values set that I could have deleted? It would seem that if there is a variable field that the file names are passed thru, it should be uneditable by a user.

eshifri, 2021-12-06:
TX is expecting model images to be in the "IMAGES" folder on the SD card. It is hard-coded location.

pfeerick, 2021-12-06:
No. As I said before, it is hard coded into the firmware to look in the `IMAGES` folder. 

It is also hard-coded that it is looking for BMP, PNG or JPG file extensions. Maximum of 10 characters to a filename. 

i.e. This is the expected structure. And the images can't be in any sub-folders either. 
[image]

If you have the correct file/folder structure, and it is not letting you pick a file at all (i.e. it comes up with the "No files on SD card!") warning I have no idea what's going on. You'll need to share some video / photos of what you are doing, screenshot of IMAGES folder, etc. btw, you said you're using TX16S, but never mentioned what firmware version?

Carl-RC, 2021-12-06:
It does not come up with any messages; in Companion, no list shows up. On the radio, I can see the image files in the IMAGE folder; but I can’t select one for a widget. Tomorrow, I will take a close look at the folders and see if I can find anything wrong. I may respond with a few pictures.

[image]<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F5500713%2F144783429-d92249d0-c8cc-424d-9cb4-3f4861ac1843.png&data=04%7C01%7C%7C597864877d0642915cd508d9b86bd7fd%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637743595588764332%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=tX8L7FBjzwrPNFLx6JaVzrtyt3YdMzzud1lWGRN5GCQ%3D&reserved=0>

eshifri, 2021-12-06:
What do you mean by "select for a widget"? 
You can select an image in the model setup. It will be shown in the widget, if you select "model.bmp",
There is no way to select widget image independently.
Or I misunderstood the sequence?

pfeerick, 2021-12-06:
Ok, lets focus on one thing at a time... there is no point messing around in Companion as you wont' be able to configure the images there if you haven't set your SD structure path up to point to a copy of your SD cards contents on your computer. It would only show you that you *have* configured a model image, if you have. Nothing more. In fact, you are more likely to have lost your model image setup this way, if you haven't setup the SD structure path. 

Focus on the radio only, and get that to work (again?). What version of EdgeTX was it working with? What version are you on now? And please don't confuse selecting the model image on the radio with it being shown on screen, as all you are doing is setting the option. After that, you need to have a widget... ie. 'ModelBmp / Model' ... to actually show the image that you configured in the model settings. Hopefully when you get a chance to look at this further we'll be able to figure out what's going on ;)

Carl-RC, 2021-12-06:
This morning I did a few things to see if I could figure out what's going on; my Radio is a RadioMaster T16s Pro Max.

  1.  I saved the entire SD card to a Temp folder on my PC so that I could have the models, image and RADIO folder files available.
  2.  I reformatted the SD card with FAT32
  3.  I copied the downloaded hourus SD card structure to the SD card.
  4.  I copied the firmware file tx16s-28cdb40.bin to the SD card FIRMWARE folder
  5.  I  copied the images from the temp location to the SD card IMAGES folder.
  6.  I copied the model files from the temp location to the SD card MODELS folder.
  7.  I copied the radio files from the temp location to the SD card RADIO folder.
  8.  I copied all the sounds from the temp location to the SD card SOUNDS folder.
  9.  I checked all the other SD card folders to see what was in them and compared them to the temp location; everything seemed OK
  10. I re-flashed the firmware in the Radio.

  *   Nothing changed when I tried to select an image file for a widget when using EdgeTX Companion.
  *   I also cannot select sound clips when I setup a PLAYTRACK when editing a model when using EdgeTX Companion.
  *   When I was editing a model (OMP M1) on the radio, I found a blank field where the model image label should be; I entered the correct image name and the model appeared in the Widget.
  *   I checked another model (Mustang) on the radio and its image was in the Widget.
  *   It appears that when creating a Widget on the Radio and selecting a ModelBMP for a Widget, you have to go back to the model and edit the image field to get the image to appear; that's what I did to get the image to appear on the radio display.

Conclusion: it appears that the Radio may in fact be OK! It appears that all the problems I'm having is when I try to edit a model using EdgeTX Companion.

  1.  I can't select an image for a model widget (I assume Companion tries to read these from the SD card?)
  2.  I can't select a sound clip for PLAYTRACK (I assume Companion tries to read these from the SD card?)
  3.  I can read the models from the SD card!
  4.  When the TX is connected thru the USB port, I can see the SD card with Windows 11 file explorer, and I can read or write to it.
  5.  Maybe I'm missing something in the Companion setup?

Carl-RC, 2021-12-06:
Since earlier this morning I have continued to explore issues with my T16S Pro Max.

  1.  I manually entered the names of the .wav file into the correct fields for switch PlayTrack entry when I was editing a model with EdgeTX Companion.  When I wrote the models back to the TX , the sound clip *.wav file reference was there with all the other model information; but, when I actuated the relevant switches, no audio was heard from the Radio?
  2.  I edited the model on the Radio and when I got to the page where I could edit the audio clip for a switch, a message popped up saying there were no sound files on the SD card.
  3.  I used the System menu to look at the SD card Sounds folder and it's filled with *.wav files.  I could select one and it would play.
  4.  It appears the firmware isn't looking in the right folder for the Sound files?

C. Ricard

Carl-RC, 2021-12-06:
Here are my model files in case you think there might be something wrong with them.

C. Ricard

[Airplanes]
model1.bin
[Helicopter]
model2.bin

Carl-RC, 2021-12-07:
Ok, success with the audio for both Companion and the TX. I was in companion trying to set the .wav audio clip and happen to click on the blue arrow beside the variable (which plays the clip, which I didn't know) and an error popped up saying it couldn't find the path to the sounds and showed me the path it was looking for. What was missing in the path was SOUNDS/en/*.wav files. When I added the en sub folder and moved all the sound files into it, all the sound stuff appeared to work in Companion and the TX. I don't know if that's how it's supposed to be installed, but it seems to work!

Carl Ricard

[image]<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F5500713%2F144783429-d92249d0-c8cc-424d-9cb4-3f4861ac1843.png&data=04%7C01%7C%7C597864877d0642915cd508d9b86bd7fd%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637743595588764332%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=tX8L7FBjzwrPNFLx6JaVzrtyt3YdMzzud1lWGRN5GCQ%3D&reserved=0>

pfeerick, 2021-12-09:
Once again, lets focus on one issue at a time, and one side at a time.

> I don't know if that's how it's supposed to be installed, but it seems to work!

As I have now stated multiple times, in order for you to be able to 'select' (and play, which you have now realised) sound files, model images, etc when using Companion, you need to have:

1. a copy of your SD card content in a folder on your computer
2. have configured Companion to know where that folder is located (`Settings -> Settings... -> SD Structure path`)

It needs that folder/files in order to be able to access the files as they would be on the radio, and configure stuff as it would be would be on the radio! If you don't have that folder, all you will end up doing is removing any image/audio configuration you did on the radio. 

So for the sounds, did you somehow have the `en` folder missing on the SD card? If so, that would explain why the radio wouldn't have any sounds (other than system sounds). 

It sounds like you've finally gotten everything properly configured so closing this issue now. Feel free to comment further if there is anything else you are stuck on. Or better yet, join us on [Discord](https://discord.gg/wF9wUKnZ6H) ;)

closed 2021-12-09 by pfeerick