EdgeTX/edgetx #1221: Companion Enhancement: On the Outputs, Logical Switches, and Special Functions tabs, make column labels stay in view.

Miami-Mike, 2021-12-06:
On Companion's Outputs, Logical Switches, and Special Functions tabs the column labels are attached to the tops of the columns so that when you scroll down they disappear off the top of the page. They should instead be attached to the top of the page so that they'll always be visible.

On [the OpenTX forum](https://github.com/opentx/opentx/issues/8788) I was told "A fix for that was tried a few years back however it affected the application performance significantly and was never progressed. The change was to use Qt table rather than Qt grid. One would assume no real difference but there was."

The problem of disappearing column labels is most apparent on the Logical Switches page, where it can be hard to remember which is the AND column, or whether the last two columns are Delay and Duration or vice versa. From the viewpoint of someone like me who's unfamiliar with the source code it seems like this behavior ought to be easy to fix, but apparently it's not.  

So perhaps you EdgeTX developers might want to take a look at this. I figure it might be worthwhile to get a second opinion.

elecpower, 2021-12-06:
As EdgeTX is using a later version of Qt I can retest to see what performance is like as the code was transferred but is disabled

elecpower, 2021-12-06:
So testing reveals still slow. Roughly doubles load time (using debug times) whether first load of a model or repeated loads of same model. Appears slightly more sluggish switching between tabs but that is just an impression (no debug times).
Also the display is ugly so work to fix that up.
Using the current grids and separating the column headings from the data causes column alignment issues that vary between OSes. Just some fun Qt thinks it knows better stuff when laying out the gui. So some coding gymnastics required.
Let's keep this open as I agree it could do with attention. But as it is an enhancement will need to wait until the big ticket items are addressed first.