EdgeTX/edgetx #1224: Tinyhawk 2 restored model bound as D8 losing bind unless Internal Radio is disabled and reenbled.

lubbet, 2021-12-06:
I updated a Taranis QX7 ACCST from OpenTX 2.2.4 to EdgeTX 2.5.0.  I used an OpenTX .otx file backup taken from the radio to restore the models and settings. My Tinyhawk 2 does not work unless the internal radio is disabled and then reenabled.  This must be done each time I power on the TX. Another user on Discord (Dan Richmond) reported the same issue (I am Dumont)

lubbet, 2021-12-07:
I just had a chance to verify the bind for my 5" drone, which uses D16.  It does not appear to have the same issue.

pfeerick, 2021-12-07:
So does the Tinyhawk 2 use D16 also, or is that D8?

lubbet, 2021-12-07:
@pfeerick The Tinyhawk 2 uses D8 as mentioned in the subject of the issue.  I believe that it only does D8. I can do some research and see if D16 is an option.

DanRichmondFPV, 2021-12-07:
D16 works fine, external R9 module works also.  Only D8 doesnt work. can get it to work by turning off the internal RF in first menu page then turning it back on then back to D8 mode and it fixes it.. also switching to one of my R9 models then back to the D8 model fixes it but switching to a different R9 model doesnt fix it (unless i do it three times in a row)  usually it doesnt work when first turning on the radio but sometimes it does i havent figured out why

DanRichmondFPV, 2021-12-07:
also QX7 ACCST

pfeerick, 2021-12-07:
Ah sorry, didn't notice it in the title 👓 ... 

Ok, so At the moment QX7 ACCST + D8 is the common denominator. Turning the Internal RF off and on again fixes it... so what about changing from D8 -> D16 -> D8? Or is a OFF -> ON cycle of the Internal RF module needed. 

What about with a new model (binding is not model specific, and with D8 you don't even have to set the receiver id number) (just to rule out a conversion issue). Unfortunately I don't seem to be able to get what I think is the only D8 receiver  I have (R86C) to bind to my X9D+ (but I can bind a D16 X8R so I know the RF board is still ok, and I can bind it on a TX16S via MPM  :-/)

DanRichmondFPV, 2021-12-07:
switching to a D16 model then back to D8 makes it not work even if it just was working

DanRichmondFPV, 2021-12-07:
creating a new D8 model automatically binds to it even tho i didnt do the bind procedure yet... still requires turning RF off and on

pfeerick, 2021-12-07:
Thanks for that... so it's probably something to do with configuring/initialising D8 then. 'Automatically binding' is understandable as with both D8 and D16 the bind is specific to the radio hardware, so once a receiver is bound in one model, it's bound for all created models. D16 just adds a receiver number that needs to match on both sides also.

Namr, 2022-01-05:
I'm having the exact same issue with a Mobula 6 bound to a D8 internal receiver on an Taranis QX7.

mricho, 2022-01-09:
Had same issue as above. using x7 ACCST, in D8 with Tinyhawk 1 and Newbeedrone Hummingbird f4 pro.
Frequently lost binding. Switched to CRSF and back, and binding was lost. The issue also occured for me on latest OpenTX, which lead me to try EdgeTX

Tried latest OpenTX, EdgeTX "Santa" 2.6.0-RC2, and last stable EdgeTX(2.5.0)

Flashed back to old TBS nightly OpenTX from when CRSFshot was released, restored models, and its working as expected

Mipme, 2022-11-18:
Running EdgeTX 2.7.1 on a QX7S ACCST with a V8FR-II I can confirm that the problem is still present... 🥺