EdgeTX/edgetx #1227: Mixer-scrippts: returning a string in output-table longer than 6 chars is fine in simulator but crashes radio

wimalopaan, 2021-12-07:
This script runs in simu but instantly crashes the radio:

```
local input = {
    {"Input", SOURCE},
    {"Reset", SOURCE},
    {"Scale", VALUE, 0, 100, 10}
 };
 
local output = { "Inc" }
 
local value = 0;
 
local function run(s, r, d)
    if (r > 0) then
       value = 0;
    else
       local i = (s * d) / 10240;
 
       value = value + i;
       value = math.min(value, 1024);
       value = math.max(value, -1024);
    end
    return value;
end
 
return {
     input = input, 
     output = output, 
     run = run
};
 
```

jfrickmann, 2021-12-08:
@raphaelcoeffic I need some help here. As far as I can see:
- In lua/interface.cpp, thel names are stored as internal Lua strings w/o any limits on the length other than Lua's max.
- In strhelpers.cpp and in gui/colorlcd/model_mixer_scripts.cpp, 16-byte strings are allocated w/o any boundary checks.
- In yaml_datastructs_x10.cpp it looks like the 6-byte limit is imposed. Is there any boundary check?

I could take it out of Lua, truncate it, and stick it back in. Or throw an error if it's too long. I don't know if a Lua error would even prevent YAML from trying to save it, and crash if it's too long?

Also, should it be 6 or 16 chars long?

wimalopaan, 2021-12-08:
Since this is only a descriptive name in the UI and plays no role as identifier, I would propose to silently cut it to 6 chars (I think 6 chars is the limits we have also in other use cases).

pfeerick, 2021-12-09:
> Also, should it be 6 or 16 chars long?

I would suggest that the 6 byte limit is from the bin file, so if 16 characters is usable everywhere else, I would bump that. And also enforce it in the mixer as I'm guess that is what crashed, and we don't want that ;)

wimalopaan, 2021-12-09:
What about the smaller radios with limited RAM? We have sometimes discussions about RAM usage in those radios.
Are those strings dynamically allocated or fixed stringbuffes?

pfeerick, 2021-12-09:
It's not really a limitation if the mixer code is already reserving 16 characters and just not using them! So if both colorlcd and B&W are the same, well, use the space reserved. Otherwise, it could be 6 (or perhaps 8 characters for B&W and something silly like 32 for colorlcd. More investigation will be needed ;)

wimalopaan, 2021-12-09:
> It's not really a limitation if the mixer code is already reserving 16 characters and just not using them! 

Maybe we could use the same length in all descriptive names? I would argue towards using (and fix) the use of descriptive names (input/lines, mixer/lines, mixer-scripts inputs/outpus, ...) with a shorter length (e.g. 8 chars) and truncating them if needed (e.g. lua output-table). Maybe this would save RAM and at the same time give more clearness.

jfrickmann, 2021-12-10:
OK, I have now chopped the names to 6 chars, as that was the breaking limit.
@wimalopaan , please check #1237 and let me know if it works as intended!

wimalopaan, 2021-12-10:
Many thanks for that.

Looks ok for me now.

closed 2021-12-13 by pfeerick