EdgeTX/edgetx #1229: Tx16S w/touch, EdgeTX 2.5 model switches update on Enter press

Avalonnw, 2021-12-08:
Hi,
Not sure if it's a bug or if I'm doing something wrong, but on model setup page when list of switches to check is highlighted, holding Enter does not do anything. On OpenTX it sets the checks to current switch positions.

eshifri, 2021-12-08:
Do you see the moving box, when the rotary encoder is rotated? This box selects the group of buttons on the screen.
When the whole switches group is selected (blue box around the group), press "enter" - the first switch will be selected. Next press will change the state of the switch: UP/---/DOWN/OFF.
Can you try it?

Avalonnw, 2021-12-08:
It works on a button-by-button selection, but not on a group. In OpenTX
holding enter button when the group is highlighted sets all the switches in
the group.

eshifri, 2021-12-08:
Aha! This functionality does not exist in EdgeTX.
Can be implemented, I suppose.

pfeerick, 2021-12-08:
There's a couple of group behaviours that aren't implemented - i.e. both pot and switch warnings gave the ability to auto-detect position of all the switches/pots/sliders by long pressing on the group.

Eldenroot, 2021-12-08:
> There's a couple of group behaviours that aren't implemented - i.e. both pot and switch warnings gave the ability to auto-detect position of all the switches/pots/sliders by long pressing on the group.

maybe again - create a list of unimplemented features which are missing from opentx due to code changes...

pfeerick, 2021-12-08:
Maybe find them first? It's easy to say "list them" when you know what they
all are... But, as is shown here, they are only being found when someone
who expects that old behaviour finally says something about it ;) And as
such, documented here now.

If it were features/behaviours that were intentionally not included or
would not be added, that would be a different matter entirely.

Avalonnw, 2021-12-08:
I was under impression, that EdgeTx was forked from OTx. Shouldn't it
transfer all behavior to Etx? It's been like that for a long time.

pfeerick, 2021-12-08:
Where it make sense, yes. But that is also like saying "just because we
once lived in caves, we should still be living in caves". It's not going to
act the same as OTX *just because*. Features and behaviours *will* be
changes from that of OTX where it is deemed necessary or makes more sense.
And when it is done deliberately, it will be documented and highlighted in
the release notes.

What you are probably missing is that if ETX has this feature missing, so
does OTX 2.4 ATM ;) The entire GUI interface was re-written from scratch,
so not every single behaviour, especially some of the not so obvious or
more specialised ones were not finished/implemented yet. So as people fine
them, if it doesn't make sense to implement it differently, we try to
re-implement it as it was.

For example, is long pressing a group an intuitive way to auto detect, or
should there be another button or control for that? Or is there some other
mechanism that can be consistently used across the whole UI. You and I may
know about that, but what about new users?

Avalonnw, 2021-12-08:
Makes perfect sense. The problem I see now is the interface is not THAT different to expect different behavior. It's like different phone brands running android: sure, there are differences but most of the stuff works the same way.

eshifri, 2021-12-09:
There is one more problem here: functionality supported with buttons may not be available with touch.
E.g. for switches we can create FormGroupExt class that will handle long press, auto-detect positions and switch warning on (if they are off) or off. There is no way to use this functionality with touch interface.

raphaelcoeffic, 2021-12-09:
The feature is implemented in #1040 (see https://github.com/EdgeTX/edgetx/pull/1040#issuecomment-989678480)

Eldenroot, 2021-12-09:
> Maybe find them first? It's easy to say "list them" when you know what they all are... But, as is shown here, they are only being found when someone who expects that old behaviour finally says [...]

I think that opentx is quite silence now... so when you forked it, it could be compared and make cherry picks :)

pfeerick, 2021-12-09:
I'll just say this again "if ETX [colorlcd] has this feature missing, so does OTX 2.4". So what do we cherrypick? 😆 2.4 OTX, which is what ETX is based on, is a complete re-write UI, so it's not like you can just compare 2.3 to 2.4 and cherrypick the difference ;) We, and you guys, need to spot the difference, let us know... hence this issue ... and then it will be addressed one way or another.

closed 2021-12-17 by raphaelcoeffic