EdgeTX/edgetx #1235: focus get lossed if and screen isnt reactive

jogybaer0815, 2021-12-09:
TX16S
edgeTX 54e7caf0  from today

if i select e.g. Model Setup, scroll down a bit and tap elsewhere focus ist lost and Scroll wheel no longer functions afterwards.
I am also not able to touch something. 
RTN,PAGE>, PAGE< Knob do not show any reaction. 
I wanted to calibrate the screen first but can't find this menu item any longer.
But the lost of focus and the impossibility to select any item via touch says something else.
All is selectable and changeable via scroll wheel.
Please have a look.
Regards

gagarinlg, 2021-12-09:
Again 😞

gagarinlg, 2021-12-09:
Does it also happen in simulator or simu?

gagarinlg, 2021-12-09:
It happens when an area which is not an input is touched and there is a widget on the screen on the last active view. The touch events on the non input areas are forwarded through the tree and the first window that is interested handles the event.
I can take a look at the weekend.

pfeerick, 2021-12-11:
It looks like a secondary aspect to this is if you manage to get back to the main screen, it plays havoc with the top bar and the entire top part of the screen appears no longer touch responsive.

jogybaer0815, 2021-12-12:
It is not possible to leave the current Menu after panel-touch:focus is deleted and only SYS or RTN reacts but only onetime and back and forth. 
Touch only allows vertical scrolling.
I have to shutdown and restart TX16S

I have just flashed the main version from October and everything goes wonderfully there.
But I'm sure you will work on it and find the bug.
Thanks for the support

pfeerick, 2021-12-12:
Yeah, the touch/libopenui whisperers are working on it ;) 

Another fun symptom... double tap on at top bar widget and watch the top bar go stupid (rather than ignore)! 😆

jogybaer0815, 2021-12-12:
I agree. But there is no escape except restart the TX16S (topscreen off).
By the way, I've now noticed that the topbar remains reactive and I can end the menu situation.
Regards

pfeerick, 2021-12-12:
The workaround *for now* is to not have any widgets in the top bar, or on the screen when you go into a menu... as it's the widgets somehow stealing touch input/focus that seems to be the cause of what we're seeing.

gagarinlg, 2021-12-12:
If you can compile yourself, go to radio/src/thirdparty/libopenui/src/button.cpp
and in onStartTouch change "return true;" to "return false;"
This fixes the issue for me, but I do not know if there any side effects, that do not happen to me. That is why I did not create a PR with this fix.

pfeerick, 2021-12-12:
Feel free to try this build out if you're brave - it has the mentioned change. 
[tx16s-onTouchStart-false.zip](https://github.com/EdgeTX/edgetx/files/7698583/tx16s-onTouchStart-false.zip)

jogybaer0815, 2021-12-12:
removing 

> The workaround _for now_ is to not have any widgets in the top bar, or on the screen when you go into a menu... as it's the widgets somehow stealing touch input/focus that seems to be the cause of what we're seeing.

that don't work for me - i will try your 'tx16s-onTouchStart-false.zip '
Thanks

pfeerick, 2021-12-13:
Fixed by dbd41d2ea959fa3d787743a9370cc99046ee6c7e

You should be able to switch back to the latest nightly now, or preferably tomorrows as another fix or two will be present there as they missed the window for todays.

closed 2021-12-13 by pfeerick