EdgeTX/edgetx #1238: Trainer 3.5mm stopped working

B-oket, 2021-12-10:
3.5mm jack not working anymore.

3.5mm jack stopped working since a few builds back.
Used to connect the 3.5mm jack to a USB dongle to run Realflight sim but the radio
is not recognized anymore, seems like there's no signal from the 3.5mm jack?

To Reproduce
Connect 3.5mm jack to USB dongle and run Realfligt (or any other sim/sw that uses 3,5mm jack)

 - EdgeTX version 2.5.0 nightly #48bd1ac
 - Radiomaster TX16S

pfeerick, 2021-12-10:
You could be suffering from issues with the 3.5mm jack (TX16S is notorious for this - it has the wrong length 4-pole socket), or it could have stopped working again. If you revert to an older build, and it starts working again, that will at least eliminate the hardware being dodgy.

tonims1, 2021-12-10:
tried the same version of nightly on a tx16s and it works for me

B-oket, 2021-12-10:
> You could be suffering from issues with the 3.5mm jack (TX16S is notorious for this - it has the wrong length 4-pole socket), or it could have stopped working again. If you revert to an older build, and it starts working again, that will at least eliminate the hardware being dodgy.

Tried another cable and now it is working again...
Strange because the cables/jack are identical but different brand..

B-oket, 2021-12-10:
Not a bug, hw issue.

closed 2021-12-10 by B-oket

pfeerick, 2021-12-10:
One is probably a fraction longer than the other, or just seats better. I've opened all of my TX16S and my TX12, and squashed the bottom two contacts a little further, so that they make contact properly with standard TRS 3.5mm connectors... as before that they wouldn't work at all, or were intermittent.

rotorman, 2021-12-10:
The signal contact on TX16S 3.5mm jack is actually the one before last (the tip-most ring of TRRS tip-ring-ring-sleeve) connection:

[image]