EdgeTX/edgetx #1239: Jumper T16 Multiprotocol EdgeTX Emergency Mode

DonaldHillNZ, 2021-12-10:
Hi, I have been trying to upgrade my older JumperT16 Pro with EdgeTX before I move to upgrading my Radiomaster T16S to EdgeTX. However Ive run into a problem I dont seem to be able to get around. My JT16 has an Internal Module that has been updated with the latest from Pascal Langer. If I leave my models with an internal module selected I get Emergency Mode all the time. Changing the models to No Module or external module it works but of course Ive no telemetry etc etc.
Ive tried everything I can think of including reloading the firmware about 3 times and reformatting the SDCard just in case.
Going back to OpentTX works fine.
What am I missing or is there a problem with EdgeTX and the Multiprotocol Module?

raphaelcoeffic, 2021-12-10:
Is that only on boot? What happens if you turn the MPM on the current model?

DonaldHillNZ, 2021-12-10:
No if I dont reboot the J16 and just exit it is ok. Once I reboot thats 
it.

MPM?

I am quite experienced and knew in OPENTX you could get this when the 
models are showing the wrong Module or no module but the only way to fix 
it is to uncheck any modules both internal and external

Mode (Issue #1239)

This email has been checked for viruses by AVG.
https://www.avg.com

raphaelcoeffic, 2021-12-10:
MPM = Multiprotocol Module

DonaldHillNZ, 2021-12-10:
HAHA of course drrr sorry. Turning on the MPM internally which it is causes Emergency Mode in the current model. Ive deleted all models and have only 1 loaded to use as a test

DonaldHillNZ, 2021-12-10:
Have tried loading a new model same result

raphaelcoeffic, 2021-12-10:
Ok, can you please post your RADIO and MODELS directory?

DonaldHillNZ, 2021-12-10:
Yes I can but if you don mind it will need to be in the morning. Its 8pm here unfortunately and im unable to go to my shed now. Should i just add it to this post tomorrow

DonaldHillNZ, 2021-12-10:
But thank you for your interest

DonaldHillNZ, 2021-12-10:
Got to fly at 7.30 pur time but Ill get it to you 11am ish my time. Very keen to get EDGETX on all my tx's

pfeerick, 2021-12-10:
@raphaelcoeffic I meant to tell you the other night, there have been a couple of reports of this the last couple of days, and it was suggested that these were early gen T16s... possibly with hardware issues that were fixes in later revisions. The posters said they had issues with 2.5 and nighties, and went back to 2.4 and had no issues there. (It's in the the #flasher discord channel, naturally)

raphaelcoeffic, 2021-12-10:
Hmm… not fun. Mine is behaving. We might try the same fix as with T18? I believe both should be very similar.

pfeerick, 2021-12-10:
It won't hurt to try. I believe Richard said somewhere he finally re-assembled his T16 Pro and was able to reproduce this.

DonaldHillNZ, 2021-12-10:
Hi Raphael,

Radio and Models files attached. Note that I removed all the models 
during my testing only leaving one but this is the full version.

My Jumper T16 is an early model that had the internal multiprotocol 
module added at a later date.

Donald

Mode (Issue #1239)

https://www.avg.com

[Models]
model1.bin
model2.bin
model3.bin
[RadiomasterBUMo]
model4.bin
model5.bin
model6.bin
model7.bin
model8.bin
model9.bin
model10.bin
model11.bin
model12.bin
model13.bin
model14.bin
model15.bin
model16.bin
model17.bin
model18.bin
model19.bin
model20.bin
model21.bin
model22.bin
model23.bin
model24.bin
model25.bin
model26.bin
model27.bin

DonaldHillNZ, 2021-12-10:
Sorry I sent the individual file but hopefully thats ok

https://www.avg.com

DonaldHillNZ, 2021-12-10:
Can I assume that the Radiomaster T16S doesnt have a problem like this?

rotorman, 2021-12-10:
@DonaldHillNZ there is no attachment in your above post. Can you please zip together the files and then drag here to a comment field.

DonaldHillNZ, 2021-12-10:
can do

DonaldHillNZ, 2021-12-10:
[MODELS.zip](https://github.com/EdgeTX/edgetx/files/7695880/MODELS.zip)
[RADIO.zip](https://github.com/EdgeTX/edgetx/files/7695882/RADIO.zip)

DonaldHillNZ, 2021-12-10:
Just off to have lunch for 20 mins

DonaldHillNZ, 2021-12-10:
back

DonaldHillNZ, 2021-12-11:
Hey guys. Ive just loaded EDGETX 2.4 and its working fine. Loading 2.5 and I get that EMERGENCY MODE as described

DonaldHillNZ, 2021-12-11:
Whilst here can you load EDGETX without the heli?

eshifri, 2021-12-11:
You need to run your own build (at this point) to remove heli. Hopefully, build-by-demand will be available soon.

DonaldHillNZ, 2021-12-11:
Is the issue in 2.5 and older Jumper t16 sets being worked on? 2.4 is ok from what I can see

DonaldHillNZ, 2021-12-11:
and are there any issues with the Radiomaster t16s pro similar to the above before I upload 2.5 to my RM

raphaelcoeffic, 2021-12-11:
> Is the issue in 2.5 and older Jumper t16 sets being worked on? 2.4 is ok from what I can see

no worry, we’ll find a way, but it might take a little time.

DonaldHillNZ, 2021-12-11:
Hi Raphael, The issue does appear to be with 2.5. The 2.4 I've loaded seems fine. Just as another test I reloaded 2.5 and same problem. I do understand the time and really appreciate the effort going into EdgeTX. So time isn't an issue at all. Look forward to migrating everything to EdgeTX

blandoon, 2021-12-12:
Hi,

I am one of the T16 users who reported an issue like this in the Discord. On EdgeTX 2.5 when I try to use MPM (by creating a new model, or duplicating one and changing its settings), it works normally at first. But if I power off and on with that model selected, it freezes on boot or at best comes up in emergency mode. Even if I select the MPM model after powering up with another one selected, I get emergency mode there too. I am stuck until I pull the battery, then it reverts to the previous model and works fine as long as I never go near the MPM one. In other words, I can get the MPM to work fine until the first time the radio is powered off, then that model can never be used again.

My T16 is a very early production unit that has been upgraded with internal MPM, Hall gimbals, USB-C, etc. Interesting to see that people with early T16s seem to have these issues.

I would love to have a solution to this - long term I plan to migrate to ELRS for most things, but for now I have a few models that require the MPM, and the T16 is the only radio I own. I can upload my models & radio files (current and OpenTX 2.3 backup) if it would help.

DonaldHillNZ, 2021-12-12:
You describe the fault perfectly blandoon. I do have version 2.4 running ok. I also want to migrate to Edge as well with both my Jumper T16 (as described in your message) and my Radiomaster T16S Pro but am loath to just yet.

pfeerick, 2021-12-13:
Your TX16S should be fine, as the fault sounds similar to that which was present in the Radioking T18, and it's a fault in the hardware implementation of the power supply for the Jumper MPM... it was mitigated in software by allowing more time for startup, and we'll probably have to do the same here for the Jumper T16 to take into account the first gen models.

richardclli, 2021-12-13:
> 

Where are the delay settings?  I can test it out with my T16, it will sometimes hangs and go into emergency mode.  I can see if the delay in TX18S works.

pfeerick, 2021-12-13:
Per this commit: https://github.com/EdgeTX/edgetx/commit/03188cd333d38f396cb304c42d610a9ea7e720f3

blandoon, 2021-12-20:
I had thought that I might be able to build and test that delay setting myself but this past week got away from me... I might get to it at some point, but in the meantime if someone else is willing/able to produce a build with that change I can test it on my T16.

pfeerick, 2021-12-20:
[T16_delay_a892cc2.zip](https://github.com/EdgeTX/edgetx/files/7743810/T16_delay_a892cc2.zip)

```diff
diff --git a/radio/src/targets/horus/board.h b/radio/src/targets/horus/board.h
index 7450b124a..24ff964b0 100644
--- a/radio/src/targets/horus/board.h
+++ b/radio/src/targets/horus/board.h
@@ -146,7 +146,7 @@ uint32_t isBootloaderStart(const uint8_t * buffer);
 void SDRAM_Init();
 
 // Pulses driver
-#if defined(RADIO_T18)
+#if defined(RADIO_T18) || defined(RADIO_T16)
 
 // TX18S Workaround (see https://github.com/EdgeTX/edgetx/issues/802):
 //   Add some delay after turning the internal module ON

```

blandoon, 2021-12-20:
So far that appears to have solved the issue for me. I got that build installed on my T16 and created (actually duplicated) a model using the MPM, flew my (already bound) D8 tiny whoop, then switched back to an ELRS model and flew that one. I powered off and on with the D8/MPM model selected, as well as switching back to it after powering on, and never got any lockups or emergency-mode boots.

Thanks so much for looking into this.

rotorman, 2021-12-20:
By looking at [Jumper T16 schematics](https://github.com/opentx/opentx/files/5035629/T16.Schematic.pdf), it is indeed similarly to [RadioKing TX18S](https://www.rcgroups.com/forums/showthread.php?3924159-Diagnosing-and-solving-system-hang-on-RadioKing-TX18S-while-turning-on-the-int-MPM), missing an inductor on the internal RF module power rail, that TX16S has (Sec. 2.11 in [schematic](https://www.rcgroups.com/forums/showthread.php?3869543-Blog-17-RadioMaster-TX16S-schematic-diagram)).

PR #1279 (2021-12-23): Add power delay for when turning on MPM for T16
Fixes #1239

Apply same power delay for T16 as TX18S, as it seems at least some (perhaps the early version?) of the T16 suffers from the same power supply instability.
commit a1c10fa4 (2021-12-23): Add power delay for when turning on MPM for T16
Fixes #1239
commit 658abdfb (2021-12-23): Add power delay for when turning on MPM for T16
Fixes #1239
closed 2021-12-23 by raphaelcoeffic