EdgeTX/edgetx #1240: Problems with log files

Salto42, 2021-12-10:
Log files cannot be read by "Lview.lua".
There is a very large production of log files; 1 for each "SD Logs" command.

Radios: T16 , TX16S

Salto42, 2021-12-10:
For the Lview.lua it's the one you're talking about.
If I rename the file as in opentx everything is fine.

pfeerick, 2021-12-10:
So the script needs updating to know the new log filename format then, or to make use of the new file listing api that the author wanted.

rboelli, 2022-02-17:
Hi, ich habe für mich eine brauchbare Lösung gefunden:
1. Bei der Datei die ich mir anschauen will, lösche ich die letzten 2 Ziffern. (das sind die Sekunden)
2. Ich habe die Lview.LUA so erweitert, das sie auch nach Stunden und Minuten sucht. (ab Zeile 493)
3. Jetzt werden mir alle ausgewählte LOG.dateien angezeigt wo ich die letzten 2 Ziffern entfernt habe, und kann sie auswählen.

Das hat den Vorteil das ich gleich einen Filter habe der mir nur die ausgewählten fils anzeigt.
Die 600 Durchläufe habe ich belassen, somit werden die files der letzten 10 Stunden angezeigt.
Das reicht für mich vollkommen aus, da mein Flugtag selten länger als 10 Stunden dauert.
Das ist ja auch nur gedacht, um sich kurz am Platz die Daten auf der Funke anzusehen, ohne ein Laptop mitzuschleppen. 
Eine genauere Analyse kann man dann ja noch am PC zu Hause machen.

gagarinlg, 2022-02-17:
Es wäre schön, wenn du auf Englisch schrieben würdest. Es sind zwar einige Leute dabei, die Deutsch können, aber bei weitem nicht alle 😉. Danke

English:
It would be nice if you would write in englisch. There are some people here that know German, but by far not all. Thanks

rboelli, 2022-02-17:
ok but my english is not so good, i tink it is better to translate with google

gagarinlg, 2022-02-17:
That's fine, you can also try deepl.com, which is sometimes better at translating.

rex600, 2022-02-22:
@rboelli
D
Hallo was genau hast du deinem Script hinzugefügt? wehre es möglich deinen code mit uns zu teilen?

E
Hello what exactly did you add to your script? would it be possible to share your code with us?

mfg

Salto42, 2022-02-22:
Hello to all,
I modified "LView.lua" as "rboelli" by extending the search to the time and renaming the files ".csv" for viewing, but recently I managed to modify it with a search by the command "dir".
I enclose the file that I modified "LView-M.lua", before launching this file it is necessary to create the folder "INDEX" under "LOGS" because all the files ".INDEX" will be saved under "INDEX". like that in the search there will be only the files which interest us.
Rename the file by deleting the ".txt".
Translated with www.DeepL.com/Translator (free version)
[LView-M.lua.txt](https://github.com/EdgeTX/edgetx/files/8117718/LView-M.lua.txt)

rex600, 2022-02-22:
Hello and fell thanks for the lua.

I installed it on my remote control and now unfortunately the display shows that no logs are available for the current model.

I created the INDEX folder under LOGs.

Do I have to do something else so that the logs are displayed to me? i think i'm doing something wrong.

Thanks in advance for your help.

Kind regards

rex600, 2022-02-22:
E Hello I just wanted to know what I'm doing wrong, didn't want to assume that the lua doesn't work.

I have a model that had telemetry and I put it on a switch under special functions safe logs, it also saves it

but I use edgetx 2.5 vill here is the error with me

I also wanted to know if the special functions are correct or if I have to call the lua in some other way.

I didn't want to accuse them of anything.

Kind regards

D Hallo Ich wollte nur wissen was ich falsch mache wollte ihnen nicht unterstellen das das lua nicht funktioniert.

ich habe ein modell welches telemetrie hadt und ich unter spezialfunktionen safe logs auf einen schalter gelegt habe diese speichert es auch 

jedoch benutze ich edgetx 2.5 vill liegt hier der fehler bei mir 

wissen wollte ich auch ob das richtig ist mit den spezialfunktionen oder ob ich das lua anderweitig aufrufen muss.

ich wollte ihnen nichts unterstellen.

Mit freundlichen grüßen

ahmad-kassem, 2022-08-26:
i have just noticed that if your model name contains 2 words separated by a space the OpenTX will not replace the space by a low dash when creating the log file and this is preventing the LView Lua from seeing the file, on the other side OpenTx will replace the space by a low dash and we will not have a problem.

i hope that EdgeTX team cam modify the software so that it adds the low dash between the words of the model name when creating the log file.