EdgeTX/edgetx #1243: In the latest Nightly c802553 version, the virtual keyboard cannot be opened and it causes a display error.

zyren, 2021-12-11:
Radio: TX16S
Firmware: Nightly c802553
In the latest Nightly c802553 version, the virtual keyboard cannot be opened and it causes a display error.

pfeerick, 2021-12-12:
Can you try again with the latest nightly... A PR which fixed the touch sensitivity also seems to have fixed this.

eshifri, 2021-12-12:
The same problem is still there (NV14).

eshifri, 2021-12-12:
Actually, the keyboard works in the mixers and Inputs tabs, but does not in the model setup.
In addition the top bar looks messed up after going between the tabs and using the virtual keyboard.
Eventually it disappears when returning to the main screen.

gagarinlg, 2021-12-12:
That is will probably to be fixed together with the touch through problem.

eshifri, 2021-12-12:
I have just rebased the code about an hour ago. This PR is already in mu build.

pfeerick, 2021-12-12:
Huh? Keyboard is working for me to edit timers, model name, flight modes, inputs names, mixers names with latest nightly, which includes the referenced PR as the only change

pfeerick, 2021-12-12:
Actually, wait, this issue is related to the TX16S, not the NV14, so whilst the NV14 has improved because #1242 is exclusively for it, this is also not going to be fixed by that PR... but I think there is another one over in libopenui which will fix it (the underlying issue is to do with widgets - if you widgets on the screen, it seems they are incorrectly grabbing the touch input).

zyren, 2021-12-12:
I tried today’s latest TX16S Nightly build d6e1213, but still when I touched any drop-down menu or edited text, the system display crashed, unable to return to the main view and the icons on the top were not touched.
This problem started from the update of #1087. I tried to comment on the sentences added in #1087 and recompile the firmware. The whole operation returned to normal.
I know this is just an attempt to find the problem, but it will not be the final solution. I just want to express my understanding of this problem. So please masters to help solve this problem, thank you very much for your hard work and express my high respect.

The screenshot below is the update I commented #1087, touch operation back to normal.

Modifications to button.h in lbopenui
[image]

Modifications to button.cpp in lbopenui
[image]

eshifri, 2021-12-12:
I have updated my code and rebuild NV14 SW - the results are the same.
On the Model Setup and Radio Setup pages ALL edit controls and Choice controls are dead.
After trying you try to use them touch return (left corner button) does not work. 
It is possible to go to other tabs and after this one can get back to the main screen.
If you mess long enough with the edit controls and tabs the top bar on the main screen is gone completely.
Sometimes you are getting the first Main Screen widget in the full-screen mode and the RTN HW button does not work.

Commenting out the code, mentioned by @zyren fixies the problem.

eshifri, 2021-12-12:
Actually, going back from 
bool Button::onTouchStart(coord_t x, coord_t y)
{
  Window::onTouchStart(x, y);
  return true;
}
to
bool Button::onTouchStart(coord_t x, coord_t y)
{
  return Window::onTouchStart(x, y);
}
also fixies it.
But I guess there was a reason for this change...

eshifri, 2021-12-12:
After the latest changes to libopenui everything works!

pfeerick, 2021-12-13:
> Actually, going back from

That would be a bit hard, since `Button::onTouchStart` never used to exist :-P

Because of a stupid decision or five in libopenui, it seems that stuff is inherited when it shouldn't be. The two devs who have been working on a lot of the libopenui changes both agree that `return true` *should* be correct, but naturally libopenui doesn't agree. It is was needed due to another function that was trying to enforce layer touch handling, but it seems to have backfired, so has been mostly reverted for now, and will be revisited in 2.7. 

Closing this issue now as dbd41d2ea959fa3d787743a9370cc99046ee6c7e does appear to have settled this down on both NV14 and TX16S

closed 2021-12-13 by pfeerick

eshifri, 2021-12-13:
OK, this means my memory failed me. :-)
I was sure at some point I saw it as a part of a commit , that was not delivered.

pfeerick, 2021-12-13:
Close, but not quite... I think it was in the discussion for another issue as a way to things work again, but that it also wasn't the right way to fix it...