EdgeTX/edgetx #1244: Unexpected behaviour when Diff and Trim are specified in the same mix

elecpower, 2021-12-11:
This mainly affects fixed wing models though from the very long discussion there are other use cases.
This was raised in OpenTX but never resolved https://github.com/opentx/opentx/issues/503
A solution was put forward but never made it https://github.com/opentx/opentx/pull/2361
Reference https://rc-soar.com/opentx/setups/diff/diff_bug.htm
I support the enhancement so I'll put if forward for the EdgeTX brain trust to consider.

eshifri, 2021-12-12:
I always thought it was a feature - not a bug. :-)
If I want the "correct" trim behavior, I can add trim as a separate line in a mix with weight, let's say 0.25 and disable trim on the stick.

elecpower, 2021-12-12:
As it currently operates, its a gotcha for a pilot who is not aware of this 'feature' especially since it is not formally documented.

One complaint about using ETX/OTX is that its too complex for newbies. Also some pilots would be happy with templates and wizards to set up and maintain their models without getting into the 'weeds'.

Another situation is where you have that brand new glider and it flies straight and true so no need for splitting the mix lines. Then after a few mishaps it needs a tweak to the trims, even after hardware adjustment. Further more we have the copy trims to sub-trims function. At that point the 'feature' raises its head. After some searching forums and issues you may stumble upon the fix.

raphaelcoeffic, 2021-12-14:
This is a quite long and historical discussion on that topic! 😳

I would suggest modeling the different options so we are clear on what we want to have at the end.

eshifri, 2021-12-14:
I agree that current behavior is counterintuitive and the examples/models provided in OTX discussions are rather convincing.
The concern is the different behavior in OTX and ETX. That being said, if it is documented and is an improvement...

gagarinlg, 2021-12-14:
If I understand that correctly, the trim is applied to the stick and then the sum is is put into diff.
I think this resembles the behavior of old radios, where the trim rotated the potentiometer. In think this is a reasonable way to implement this.
Did I miss something?

eshifri, 2021-12-14:
@gagarinlg There was a long discussion here:  opentx/opentx#503
One cam apply the same argument to the expo, right? But in the case of expo we definitely want to exclude trim.
In some sense, differential is similar: you wan to have different sensitivity to the stick in opposite directions.
So one can find argument pro and contra both ways of doing it.
With the current behavior, if one wants to have a proper differential ailerons, he has to use trim as a setup line in the mix. Not a big deal IMO, if the behavior is known and understood. Unfortunately, many pilots find it out by learning on their own mistakes.