EdgeTX/edgetx #1249: Make HELI tab a model specific setting (instead of a compile option)

rotorman, 2021-12-13:
Instead of having to issue -DHELI=YES / NO at build time for the CMake line, it would be better to have a model specific option, where the heli tab could be enabled (default off).

pfeerick, 2021-12-14:
Not just Heli, quite a few of the tabs would be good to have that way (model level configurable)... Some flyers don't want flight modes, gvars, etc. In the first instance, it would simple be a matter or adding removing the tabs. Second instance would be removing other UI elements as required. 

This is another setting that  would be nice to have at both the radio and model level, or perhaps, better still, have a 'model default' concept...

gagarinlg, 2021-12-14:
I like the 'model default' idea.
Maybe two settings
1. advanced mode, with gvars, flight modes, enable heli tab switch
2. Heli tab enable

Then we have a cleaner UI for simple models and do not generate too many options.

pfeerick, 2021-12-14:
I'm starting to think more page/tab where you can setup some model defaults and such at the radio setup level, and then have another tab on the model setup level that is basically the same, but can over-ride anything set at the radio setup level... thus defaults AND model specific settings. This is also where some other model concepts like themes per model and backgrounds per model could be introduced... making it a UI/UX focused tab. Tickboxes for hideable tabs, and the theme/background options. One idea anyway.

kevinkoenig, 2021-12-14:
That idea makes a lot of sense.  As a fairly new user, I can confirm that the learning curve is really high.  There is a lot of cognitive overload.  An easy mode would have been very useful.  Setting the available tabs in radio settings and then overriding them in the model settings is a good way to reduce the learning curve.  Getting the defaults right will be important.

wimalopaan, 2021-12-15:
I think we still need to have the compile-time-option, because flash is starting to become a concern on smaller radios. 
So, don't remove the compile-time flag, but make it also selectable at runtime if compile-time flag is true.

Eldenroot, 2022-10-08:
> I like the 'model default' idea. Maybe two settings
> 
> 1. advanced mode, with gvars, flight modes, enable heli tab switch
> 2. Heli tab enable
> 
> Then we have a cleaner UI for simple models and do not generate too many options.

This is a good idea!

pfeerick, 2023-05-22:
Implemented by https://github.com/EdgeTX/edgetx/pull/3484

closed 2023-05-22 by pfeerick