EdgeTX/edgetx #1250: Problem compiling for Radiomaster TX16S

janmj, 2021-12-14:
Trying to compile latest from master. Done it before, but now i suddently get this error:

/home/janmj/storage/disk_1/Develop/edgetx/edgetx_main/radio/src/storage/sdcard_raw.cpp: In function 'const char* readModel(const char*, uint8_t*, uint32_t)':
/home/janmj/storage/disk_1/Develop/edgetx/edgetx_main/radio/src/storage/sdcard_raw.cpp:147:25: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]
  147 |     convertBinModelData(filename, version);
      |                         ^~~~~~~~
      |                         |
      |                         const char*
In file included from /home/janmj/storage/disk_1/Develop/edgetx/edgetx_main/radio/src/storage/sdcard_raw.cpp:24:
/home/janmj/storage/disk_1/Develop/edgetx/edgetx_main/radio/src/storage/conversions/conversions.h:24:40: note:   initializing argument 1 of 'const char* convertBinModelData(char*, int)'
   24 | const char* convertBinModelData(char * filename, int version);
      |                                 ~~~~~~~^~~~~~~~
[ 61%] Building CXX object radio/src/CMakeFiles/firmware.dir/storage/conversions/yaml/yaml_datastructs.cpp.o

This is on Ubuntu 20.04 with recommended setup for compiling EdgeTX.
The build command:
cmake -DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=2 -DGVARS=YES -DPPM_UNIT=METRIC -DHELI=YES -DLUA=YES -DINTERNAL_GPS=NO -DLOG_TELEMETRY=YES -DMODULE_PROTOCOL_FLEX=YES -DCMAKE_BUILD_TYPE=Release ../

pfeerick, 2021-12-15:
`-DMODULE_PROTOCOL_FLEX=YES` is no longer a supported option (it's on by default). However, that won't cause a build error, just this warning from cmake:

```
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    MODULE_PROTOCOL_FLEX
```

I would delete and recreate your build directory, ensure that you have `git fetch && git pull` on the `main` branch, and that you run `git submodule update --init --recursive` to make sure the submodules are up-to-date also, and then run cmake and make again and then let us know the outcome.

janmj, 2021-12-15:
@pfeerick Thank you for your input, i have already done fetch and pull, but i usually just done a make clean, so now i also removed and made a new build directory. Så now i get a totally new error :) And i'm definitely on the main branch..

/home/janmj/storage/disk_1/Develop/edgetx/edgetx_main/radio/src/opentx.cpp:740:4: warning: #warning "KEYSTUCK Message Not Yet Implemented" [-Wcpp]
  740 |   #warning "KEYSTUCK Message Not Yet Implemented"
      |    ^~~~~~~
[ 67%] Building CXX object radio/src/CMakeFiles/firmware.dir/strhelpers.cpp.o
/home/janmj/storage/disk_1/Develop/edgetx/edgetx_main/radio/src/opentx.cpp: In function 'void checkBacklight()':
/home/janmj/storage/disk_1/Develop/edgetx/edgetx_main/radio/src/opentx.cpp:549:33: error: 'class MainWindow' has no member named 'touchEventOccured'
  549 |     if (MainWindow::instance()->touchEventOccured() && (g_eeGeneral.backlightMode & e_backlight_mode_keys)) {
      |                                 ^~~~~~~~~~~~~~~~~
[ 67%] Building CXX object radio/src/CMakeFiles/firmware.dir/switches.cpp.o
make[3]: *** [radio/src/CMakeFiles/firmware.dir/build.make:1883: radio/src/CMakeFiles/firmware.dir/opentx.cpp.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/Makefile2:696: radio/src/CMakeFiles/firmware.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:703: radio/src/CMakeFiles/firmware.dir/rule] Error 2
make: *** [Makefile:251: firmware] Error 2

gagarinlg, 2021-12-15:
You have an old version of libopenui.

janmj, 2021-12-15:
@gagarinlg I'm an idiot! i forgot to run git submodule.... 

Now it compiles fin again, thanks guys!

closed 2021-12-15 by janmj