EdgeTX/edgetx #1252: FrSky X10(S) Express + MPM some protocols (MPX and maybe others) missing.

nostromo123, 2021-12-15:
Radio FrSky X10S Express with EDGE v2.5 and nightly.

MPM module Jumper 4in1 clone with CKS32.
- has been flashed with flash-multi-0.6.1 and mm-stm-serial-taer-v1.3.3.0.bin
- CKS has been verified with STM32CubeProgrammer as STM32
- MPM working under OTX 2.3.14 and showing MPX protocol.

Issue is, that under EDGE MPX protocol not showing up, when scrolling protocols on radio.
(Made a list of visible protocols)

[EDGE nightly.txt](https://github.com/EdgeTX/edgetx/files/7720294/EDGE.nightly.txt)

In EDGE companion 2.6.0 MPX is available and can be saved and transferred to the radio.
(screenshot of protocols in companion)

[image]

But when starting the radio, the message 'SCANNING PROTOCOLS' appears and no MPX shows up, when scrolling for the MPX protocol.

I posted in RCG (https://www.rcgroups.com/forums/showpost.php?p=48327043&postcount=1520)
hoping someone there can confirm or decline, but no answer till now.

In (https://frsky-forum.de/index.php?thread/6130-seit-kurzen-probleme-mit-otx-und-multimodul/&pageNo=1)
one user tested a MPM with his TX16S and EDGE v2.5 and nightly and had no problem with the MPX protocol.

br KH

raphaelcoeffic, 2021-12-16:
The protocols offered on the radio are listed from the MPM itself. So the bigger question is: does your specific MPM and it’s firmware support this protocol?

please note that Companion has no idea of the protocols really supported, it just lists them all.

raphaelcoeffic, 2021-12-16:
Sorry, it seems I cannot reproduce this. All the protos including those without sub-proto are included when testing with TX16S and MPM 1.3.3.0 (internal as well as external module). It is highly improbable it would be different with a X10, as there is nothing dependent on the radio type itself here.

raphaelcoeffic, 2021-12-16:
@nostromo123 after checking the screenshots on RCG, it seems your module is not talking to the handset, so that the dynamic protocol list is probably not used, and thus the protocol is missing.

nostromo123, 2021-12-16:
I think now, that I might have a crappy hardware in form of that MPM module here.That is the second MPM module, I ordered from China in the last two month. First one could not be flashed with the MPM firmware at all and went back. This one here now is a clone Jumper with a CKS32. Looked at first sight trustworthy. But seeing now also some erratic behavior under OTX, I'll close this here now.

Sorry for troubling you. 

br KH

nostromo123, 2022-02-05:
Have another MPM module her on my table.IRangeX IRX4.

Now my X10S Express with edgeTX 2.6

rotorman, 2022-02-05:
Please provide further info - your last message is not clear what you want to say.

Is the problem with missing MPX still there also with IRX4 ? Did you test it with X10S Express **and** TX16S and both behave similarly?

nostromo123, 2022-02-05:
@rotorman 

Hi, I don't have a TX16S. Doing those experiment with my X10 S Express.

br

nostromo123, 2022-02-05:
One step further ahead.

I did **not** update that new MPM module, I have now here on my desk.

I flashed my X9D+ from OTX 2-2-4 to edgeTX 2.6

[image]
[image]

Then I scrolled for M-Link! 

[image]
[image]

br KH

pfeerick, 2022-02-07:
> raphael wrote ' it seems your module is not talking to the handset, so that the dynamic protocol list is probably not used, and thus the protocol is missing'

This was in relation to the screenshots you posted on RCG, as they both indicated 'No MULTI_TELEMETRY detected' which is an indication that the MPM module isn't even talking to the transmitter, so the radio firmware falls back to an internal list, and the EdgeTX internal protocol list is older than the one OpenTX is using (same list is used for B&W and colorlcd radios, does not have M Link as an option). However, when the MPM is talking to the radio, the radio can interrogate it for the list of support protocols that the specific firmware version the MPM is running supports, and options for each. 

On my TX16S with ETX 2.6, and 1.3.3.0 MPM firmware, M-Link shows up as an option. If I tell it have a MPM in the external bay (and there is none), the protocol list there does not show M-Link. i.e. as per your initial result, and is expected if the MPM is not working properly. Hopefully the fact your X9D+ is showing the M-Link entry means your X10 will also work with that particular MPM module, since this has come from the module. 

Are you able to update us any further as to status of this? If you do update the MPM firmware, you should also be able to revert since you can still download older versions of the MPM firmware, including the 1.3.2.61 it is currently running. So you should also be able to identify if something changed with a newer MPM version safely. I wouldn't think that is the case as I don't see any mention in the release notes for the MPM but can't discount the possibility.

nostromo123, 2022-02-07:
Hi Peter.

- I have a back-up of the MPM firmware.

- I'll update the MPM to Pascals latest fw version 1.3.3.7
- I'll check with my X9Ds here under EdgeTX 2.6 MPX plus binding plus telemetry
and crosscheck with my X10S Express under EdgeTX 2.6 and come back.

but this week I am totally busy, maybe oncoming  weekend I can find some time.

But there is one other issue X10S Express <> MPM, this time OTX 2.3.14 with MPM
with reduced protocols ....

See https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/648

I did not test that under EdgeTX (MPX not showing!)

So I think, there could possibly be a **general** communication problem X10 **Express** <> MPM??

br KH

pfeerick, 2022-02-07:
Oh, interesting... yes, that post would certainly suggest there is something fishy going on in the the X10, especially if you can use your X9D as a reference on what the X10 should be showing ;) Since if you can't update the firmware via the transmitter, and have the NO MULTI_TELEMETRY on the X10 then indeed the module and the X10 are not talking to each other, which is a bad start. 

No hurry, was just wondering on status of the issue.

nostromo123, 2022-02-12:
Sorry for coming back here so late.

My radios X9D+ and X10S Express. OS on both radios EdgeTX 2.6.

MPM IrangeX IRX4 flashed with Pascals latest stable 1.3.3.7 TAER

**Test sequence on X9D+**

**M-Link selected and binding MPX RX**

[image]

**Telemetry data receiving**

[image]

**Wrinting the MPM fw from radio to module**

[image]

Could not take screenshots from here, so I switched to my mobile.

[image]

[image]
[image]

So far, so good!

**Now that procedure on the X10S Express**

**NO M-LINK showing up**

[image]

**NO information on the radio info screen**

[image]

**Trying to flash the MPM from the radio**

[image]

[image]

[image]

[image]

Tomorrow I'll try to do a run with the X10S Express/MPM and one of my FrSky receivers and post the result here ....

br KH

nostromo123, 2022-02-13:
X10S Express/MPM and one FrSky D8R-IIplus

**Binding and settings**

[image]

**Telemetry data**

[image]

**Servo on CH 4 and working**

[image]
[image]

No issue with that!

nostromo123, 2022-02-14:
> But there is one other issue X10S Express <> MPM, this time OTX 2.3.14 with MPM with reduced protocols ....
> 
> See [pascallanger/DIY-Multiprotocol-TX-Module#648](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/648)

Just checked that point.

Modified Pascals 1.3.3.7 (the _Config.h file by commenting unwanted protocols) compiled it in Arduino IDE and uploaded it to the MPM and with my X9D+ this is working!

cu KH

pfeerick, 2022-02-14:
Nice! Something weird is still going on though since you had it working with the X9D+ but not the X10, as well as the inability to update the firmware on the X10. Normally the answer would be 'bootloader isn't installed' but then you wouldn't be able to have update it on the X9D+ either then. 

As far as the module information screen, that seems _normal_, as the same is the case with the internal MPM with the TX16s. Would probably be better if it reported that it was MPM + version number or something, but not atm. 

What MPM did you get... is it the iRangeX IRX4 with the yellow knob on the back? I finally gave in and ordered one... have been meaning to get one for ages to use with my older transmitters.

nostromo123, 2022-02-15:
That one

[image]
[image]

br KH

nostromo123, 2022-02-15:
The only issue I could spot with that MPM is, that it has an APM32, but you'll not get a MPM with a STM 32 from China now.

Tested this module with STM32 Cube and is recognized as a STM32. So this should not matter ...

[image]

pfeerick, 2022-02-15:
Ah, that probably explains the cryptic message about the one I ordered not having a STM32 but instead an "ARM CORTEX-M" .. but, er, that is what the STM32 is? lol I'm pretty sure it should be the same as the one you have, just with most of the missing components populated (buttons, knob, leds, USB port), so will at least be able to have a point of reference using the TX16S when it finally ships/arrives. As well as a use for one of my older transmitters again ;)

pfeerick, 2022-03-12:
I received my "iRangeX IRX4 Plus" yesterday, so I can now hopefully add some more datapoints to help or confuse even more :grin: This is with RM TX16S. 

The module shipped with v1.2.0.22 firmware, and as expected, it didn't list M-Link. 
[image]
[image]

Upgrading the bootloader and firmware using Flash Multi v0.6.1, to v1.3.3.7 (as v1.3.3.0 is the version that introduced the protocol/subprotocol dynamic scan facility EdgeTX uses) and M-Link is now in the menu:

[image]
[image]

And just for reference, this is the internal MPM, which is still running v1.3.3.0:
[image]

And, on X9D+2019:
[image]

I'll open it up later to see what's actually in it.

nostromo123, 2022-03-12:
But not here with my X10S Express ..

With 1.3.3.7 TAER no M-Link showing up. With 2.6 release and with nightly.

Strange it is.

br

pfeerick, 2022-03-13:
Ok, I've just thought of something "simple" we can try... what does the protocol list look like with and without the module inserted? You'll need to power cycle between module in and out state to ensure the dynamic protocol scan happens...  i.e. Without the module it should incorrectly have "verted" as the last protocol...

[image]

... whereas with the  module it's ZSX and you can see both the Wfly entries in order. If you don't, probably means that on the X10S the dynamic protocol scanning is not working (properly). 

[image]

For reference, this is the module I have, basically seems to be the same as yours:
[image]
[image]
[image]
[image]

nostromo123, 2022-03-13:
Your MPM seems to be identical, just missing some hardware ...

Did the short test as proposed.

No MPM inserted, ext. module select

[image]

Scroll to last line in protocol selection

[image]

Radio off, insert MPM, on again, select MPM in ext. module.

[image]

and the last line in protocols

[image]

In my eyes, that looks like a specific issue with those X10(S) Express radios!?

br KH

pfeerick, 2022-03-13:
Damn, it looks like it. From the looks of yours it seems to be using the internal list, so the protocol scanning isn't working properly... I'll have to poke Raphael after 2.7 is out as I'm pretty sure he has X10 hardware so should be able to figure it out.

nostromo123, 2022-03-13:
... and as under OTX 2.3.14 I do see M-Link, that issue is not a hardware defect of my radio here. I am quite sure.

Thank you for being so persistent ;-)

pfeerick, 2022-03-13:
OTX has been maintaining the internal list of protocols for the MPM, which probably explains why you see it there. We instead opted to use the dynamic scan, which would make it so there would not be a need to keep updating ETX firmware... but obviously there is no a problem if it the two are not talking properly on this.

pfeerick, 2022-03-21:
Are you able to try a build from this PR? https://github.com/EdgeTX/edgetx/pull/1705 I updated the static list of protocols and would be curious to know if MLink now "works"... meaning we still need to find out why the dynamic protocol scan is not working, but because its in the default list it at least shows up now when scanning doesn't happen.

nostromo123, 2022-03-21:
It is not enclosed in the current nightly, I guess? So I think, I have to load the nightly source and exchange some files there and then build. Normally no problem, but at the moment it is not clear to me, which to replace in the source ....

Or modify 2.6 source and test with that?

pfeerick, 2022-03-21:
No, because it's not been merged yet. But all PRs are built automatically still (via the Checks tab at the top of the PR). 

Anyway, link to a zip file with all the builds will be at the bottom of this page : https://github.com/EdgeTX/edgetx/actions/runs/2008365108

nostromo123, 2022-03-21:
OK. ;-)

[image]

[image]

[image]

pfeerick, 2022-03-21:
By rights, it should show M-Link regardless of whatever the MPM is inserted or not, as I still don't think scanning is working properly, but because M-Link is in the fallback list now, it should show up (and hopefully work)... The giveaway for me atm is that if the MPM scanning is working, Assan is the first protocol listed , otherwise some some other protocol starting with B is the first one... Thanks for confirming that part at least is working for you! :)

nostromo123, 2022-03-21:
Anything else, I can do/test/try at the moment?

br KH

pfeerick, 2022-03-21:
Not just yet... In a week or two once 2.7 is out of the way I'll poke Raphael to see what he can make of it...

donkeydog67, 2022-12-22:
Any updates about the issue?

raphaelcoeffic, 2022-12-28:
This is a very old issue, is that even still an issue?

nostromo123, 2022-12-28:
OK, then I'll close it. Think, the issue has been solved.

br KH

closed 2022-12-28 by nostromo123