EdgeTX/edgetx #1253: LUA-Api: enumerate all mixer-scripts

wimalopaan, 2021-12-15:
It would be very helpful to have the possiblility to get a table with all lua-scripts, their names and outputs.
The outputs should list the names and the id's, and the id's should be useable to insert a mixer-line.

jfrickmann, 2021-12-15:
Instead of adding this custome functionality to EdgeTX, you could also just do it in Lua. Pick a name for a global variable, and have the scripts set it to true when they are loaded. Then the script that you want to test if they are loaded, can just look for that global variable.

wimalopaan, 2021-12-16:
> Instead of adding this custome functionality to EdgeTX, you could also just do it in Lua. Pick a name for a global variable, and have the scripts set it to true when they are loaded. Then the script that you want to test if they are loaded, can just look for that global variable.

This then only works on B&W for telemetry and mixer-script. But on colorLCD mixer-script and widget can only exchange data via  GVs or Shms. Here for every script of question a bit can be set. 

The script maybe a variable number of outputs. This too can be coded bitwise  a global var, a GV or Shm.

But how do I get the script internal id to insert that into a mixer-table as source for model.insertMix()?

jfrickmann, 2021-12-16:
I assume that each mixer script is unique, and you can define what bits it should set. I think that you are on the right track.
I just feel that this request is for some very specialized and narrow functionality that very few people would use, and given the fact that we are apparently filling up the flash memory on some targets, we need to be more selective with what we add, because there is no limit to what we _could_ do ...

wimalopaan, 2021-12-17:
Thank you for your comment!
Surely we have to be very selective on what we include. In this special case it looks like there is a workarounf using a naming-convention for the scripts/outputs and getFieldInfo().