EdgeTX/edgetx #1271: Fix KEYSTUCK for color LCD radiso

wimalopaan, 2021-12-18:
Please fix this:

https://github.com/EdgeTX/edgetx/blob/b3e85e8444e525869a75a35dfa475c4b70d5984f/radio/src/opentx.cpp#L729

pfeerick, 2021-12-19:
You mean implement, don't you? As there is nothing to 'fix' - the warning is there as a reminder that at some point it needs to be implemented or dumped so serves it's purpose.

wimalopaan, 2021-12-19:
Well, I don't know if a hole that has a clear sign on it, that there is something mssing, gets repaired or new constructed. But that might go into phylosophy ...

The interesting part is, that I had a TX16s with an stuck button on the rollerboard, and OpenTx 2.3.14 (maybe older) prints a message "Taste klemmt" ( german for: "Key gets stuck"). Looks like there is/was an fix/implementation for this.

pfeerick, 2021-12-19:
I doubt it... the warning will be there because it hasn't been implemented the libopenui way ;)

The warning is there so you don't fall in the hole... if you move the sign, and fall in the hole, well, that's on you 😆 

Either way, at least there is an issue to go with the warning now ;)

wimalopaan, 2021-12-19:
Look at: https://github.com/opentx/opentx/blob/a621b3efb83908428bf2a028b1029949059a4594/radio/src/opentx.cpp#L1033-L1040

pfeerick, 2021-12-19:
That is a starting point, but I'm afraid that is all it is... you can't just copy from 2.3 to 2.4 or ETX, because the entire UI backend was re-written. Hence that warning/reminder, and all the UI glitches and incomplete features still being finished.

wimalopaan, 2021-12-19:
> That is a starting point, but I'm afraid that is all it is.

Yes, I know that. Just wanted to point out, that this feature is/was working in OpenTx. And with some crap rollerboards from RM it was a blessing to have it in place ...

eshifri, 2021-12-19:
In the current implementation a stack key is intercepted by "Switch warning".

PR #1289 (2022-01-09): Stuck key test for COLORLCD #1271
Fixes #1271 

Summary of changes: During the boot check for stuck key and display FullScreenDialog with the warning and the name of the key that causes the problem.
The message is displayed for ~5 seconds and goes away (as for b/w TX) or can be cleared by RTN or ENTER (or touch).
closed 2022-01-09 by pfeerick