EdgeTX/edgetx #1272: Add option to invert serial rx/tx line of tx16s serial ports (and other radios with usabel serial interfaces)

wimalopaan, 2021-12-18:
The serial ports at the bottom of a TX16s use the lines uninverted. This make it easier for some DIY gear to be connected but requires using an SBus inverter if one wants to connect a receiver with "normal" SBus-Output (inverted).
Here an option to invert the serial rx and/or tx line would be cool.

gagarinlg, 2021-12-18:
As the MCU hardware does not support this, a software Uart had to be used then. This has been discussed and may be done in the future.

wimalopaan, 2021-12-18:
> As the MCU hardware does not support this, a software Uart had to be used then

Really. Sorry, my understandung of the STM32 is limited: this is very sad. 
But in fact my favorite AVRs do support this ;-)

raphaelcoeffic, 2021-12-18:
> > As the MCU hardware does not support this, a software Uart had to be used then
> 
> Really. Sorry, my understandung of the STM32 is limited: this is very sad. But in fact my favorite AVRs do support this ;-)

Interresting: based on software serial, or based on hardware support? I don't think the 8bit Atmels do have support for hardware based inversion, do they?

wimalopaan, 2021-12-18:
The newer AVR-Tiny (core1 and 2) and  -mega are assential XMega devices and they support inverting all pins including rx/tx of the usarts as well as OpenDrain-Mode for the tx-pins. Another neat feature is the muxing of most internal peripherals. For 8-bit they are real cool devices.

pfeerick, 2021-12-19:
lol... these are the new ones though... either last generation Atmel or Microchip era ones... which aren't in transmitters so don't count :-P Certainly a nice improvement though... especially the pin muxing so not restricted to this or that pin as much.

wimalopaan, 2021-12-19:
Yes, the newer STMs are cool stuff, esprecially those with some nifty coprocessors, e.g. CORDIC, etc. if you have to do some control algorithms ...
But the MicrChip-era AVRs (those I meant above) are also very cool, clearly not the power of an 600MHz-Quad-Core-ARM, but nowadays capable of going up to 32MHz.