EdgeTX/edgetx #1273: X9D (Not Plus) Haptic Motor Functionality

techfixpros, 2021-12-18:
Original X9D
ETx 2.5.0 x9d-28cdb40.bin

Please add haptic motor functionality as is possible with OTx. I have an original B01 X9D with the haptic motor installed, and OTx supports this radio with the motor.

raphaelcoeffic, 2021-12-18:
I believe this a compilation option. What happens if we enable it, even if the hardware is not present? Are there any downsides to do so?

pfeerick, 2021-12-19:
It is - `HAPTIC`

I suspect it just makes the firmware a bit bigger... since it enables the haptic stuff... it's only a motor so not like it has to do anything more than turn the power on and off. 

[image]

pfeerick, 2021-12-19:
Does this work for you? [X9D_HAPTIC_a4c718f.zip](https://github.com/EdgeTX/edgetx/files/7740144/X9D_HAPTIC_a4c718f.zip)

techfixpros, 2021-12-19:
Flashed and works properly, thank you.

techfixpros, 2022-01-26:
Instructions for custom build:

https://gitpod.io/#https://github.com/edgetx/edgetx/tree/main

export PATH="~/edgetx/radio/util:$PATH"

cmake -DPCB=X9D -DDEFAULT_MODE=2 -DPPM_UNIT=US -DHELI=NO -DLUA=YES -DLUA_MIXER=YES -DHAPTIC=YES -DCMAKE_BUILD_TYPE=Release ../

make -j`nproc` firmware

closed 2022-01-26 by techfixpros

techfixpros, 2022-09-10:
Merged in https://github.com/EdgeTX/edgetx/commit/63d4cb7848604bd55338c79fb6c5e07c27883191