EdgeTX/edgetx #1280: Simulator Crashes when Using Theme Editor

richardclli, 2021-12-21:
**Describe the bug**
Use simulator to edit the Theme with build in theme editor will cause buffer overflow error and force quit the simulator.

**To Reproduce**
1. Run simulator
2. Go to Theme editor
3. Select a theme and choose Edit
4. Crash

**General information**
 - EdgeTX version 2.6.0-RC1
 - Hardware TX16S simulation in Linux

pfeerick, 2021-12-21:
Oh, *interesting*... I used to have this happen with `simu` for a while, but then it seemed to stop... seemed somehow related to the themes loaded as it went away when I cleaned out my `THEMES`. Whereas Kevin didn't have any issues at all (which is unsuprising, since he was able to make it all work :grin:).

kevinkoenig, 2021-12-21:
Do we have a stack trace at all?

From: Peter Feerick
Sent: Monday, December 20, 2021 11:58 PM
To: EdgeTX/edgetx
Cc: Subscribed
Subject: Re: [EdgeTX/edgetx] Simulator Crashes when Using Theme Editor (Issue#1280)

Oh, interesting... I used to have this happen with simu for a while, but then it seemed to stop... seemed somehow related to the themes loaded as it went away when I cleaned out my THEMES. Whereas Kevin didn't have any issues at all (which is unsuprising, since he was able to make it all work 😁).

richardclli, 2023-02-24:
@pfeerick The problem gone in 2.8.0, can close.

pfeerick, 2023-02-24:
I would play with this a little more before being so sure.... it is a somewhat *random* error as sometimes it crashes first time, other times you need to do it a couple times in a row to make it trigger... I even had it crash or not crash depending on themes present... but fingers crossed it is finally gone! 🤞

closed 2023-05-05 by pfeerick