EdgeTX/edgetx #1283: Companion26 does not convert any models for Taranis X7-Access and X9D+2019, works for TX16S

bug400, 2021-12-22:
**Describe the bug**
Companion26 (commit cdeb92a) does not convert model data from an otx file for the Taranis X7-Access and X9D+2019 simulator. The TX16S simulator converts the model and radio configuration properly and stores them in a RADIO and MODELS directory within the LINUX /tmp directory. The converted models are available in the TX16S simulator.

The Taranis emulator creates a RADIO and MODELS directory in the sdcard directory. After the conversion there is a radio.yml file in the RADIO directory and a only singele model00.yml file with default model data in the MODELS directory.

The console report of Companion26 shows that "simuSettingsDirectory" is empty for the Taranis simulator. For the TX16S simulator this item contains the temporary file path for the RADIO and MODELS directory.

**To Reproduce**
Recreate SD-Card directory from the taranis-x7.zip file. Start companion26. Select X7 Access Radio Profile.Open an otxfile with several models (created with opentx 2.3) and start simulating the transmitter. After having confirmed the "Storage Warning" message
the transmitter shows that an EEPROM conversion is being executed. The the message "Bad Radio Data" is displayed.
After confirming that message no models from the otx file are available in the X7 Access simulator.

**Expected behavior**
Models are converted and are available in the Taranis simulator.

**Screenshots**
[qx7.log](https://github.com/EdgeTX/edgetx/files/7762804/qx7.log)
[tx16.log](https://github.com/EdgeTX/edgetx/files/7762808/tx16.log)

**General information**
 - Companion/Simulator Branch 2.6 commit cdeb92a

raphaelcoeffic, 2021-12-23:
@elecpower It seems the `.otx` file has categorised storage (`RADIO/radio.bin`, `RADIO/models.txt`, etc), and tries to start the simulator widget with these files. Obviously, the simulator widget expects either an EEPROM file or YAML files (not supported in that Companion version).

So the question is: why does Companion thinks it should not output the models / radio settings as an EEPROM file? (as was probably the case for previous versions)

elecpower, 2021-12-23:
This is one of those 'down the rabbit hole' investigations but on a superficial dig it appears that Companion passes all settings via a memory buffer and sets up paths it is told each radio type expects.
I suspect the libsims use the memory buffer to start but any writes are under control of the radio code.
This would explain multiple sd folders for b&w.
I'll see if I can make a pre-yaml interim hack that will replicate colorlcd.
I need to make sure it doesn't break the standalone version.
No promises

closed 2022-01-02 by raphaelcoeffic