EdgeTX/edgetx #1284: Curve-Editor does not immediately update x-coordinates

wimalopaan, 2021-12-22:
If one creates a new x/y-curve (e.g. 5-points),  it is clearly not possible to move x2 from -50 to say +10, because x3 is 0, and the curve must remain a mathematical function.

But if yoiu move x3 to say 20, it should be possible to move x2 to +10. But this is not possible, x2 is stuck at the last value of x3 (here: 0). One hast to leave the curve-editor and enter it again. Then it works as expected.
Looks like the give values are not propagated correctly to the curve coordinates.

gagarinlg, 2023-02-26:
@philmoz is this fixed with your changes to the curve edit screen?

philmoz, 2023-02-27:
Should be fixed in 2.8.1

pfeerick, 2023-02-27:
#2972 ended up as a duplicate of this, and fixed in #2973, which will be in 2.9

closed 2023-02-27 by pfeerick