EdgeTX/edgetx #1285: enhance getFieldInfo() to get id as well as names

offer-shmuely, 2021-12-22:
I like to write a new gauge widget
something like this:

[image]

we need  a way to read a source parameter in widget options (supplied by the user)
 and display it's info

I declare the following for user input:
```
local options = {
     {  "Source1", SOURCE, 123 }    -- RSSI
}
```

to use it:
```
// source1 is a number
fieldNameAsInt = wgt.options.Source1

// this one can not work
fieldinfo = getFieldInfo(fieldNameAsInt)
```

I love to have one of the two:
**option 1**
support int as well 
`getFieldInfo(fieldName**AsInt**) or getFieldInfo**2**(fieldNameAsInt)
`
**option2** 

implement new function
`fieldNameAsString = fieldIdToFieldName(fieldNameAsInt)`

offer-shmuely, 2021-12-23:
adding sample file

```
local options = {
  { "Source", SOURCE, 251 } -- RSSI
}

local function create(zone, options)
  local wgt = { zone = zone, options = options }
  return wgt
end

local function update(wgt, options)
  if (wgt == nil) then return end
  wgt.options = options
end

function refresh(wgt)
  if (wgt == nil) then return end

  print("wgt.source: " .. wgt.options.Source)

  -- can work with name or id
  local fieldValue = getValue(wgt.options.Source)
  print("fieldValue: " .. fieldValue)

  -- can work with name, can NOT work with id
  local fieldinfo = getFieldInfo(wgt.options.Source)
  if fieldinfo == nil then
    print("error, field-info NOT available")
    return
  end

  print("fieldinfo.id: " .. fieldinfo.id)

end

return { name = "fieldInfo0", options = options, create = create, update = update, refresh = refresh }

```

jfrickmann, 2021-12-23:
`getFieldInfo` uses tables with source names that are generated by a Python script during the build. These are similar but not the same as the names that you see on the screen when you pick a source for something:
[image]
In #1154, I have made functions for getting switch positions, and instead of creating my own names, I have used the exact same names that you see on the screen. The advantage of this approach is that it is much simpler to implement, as I use the existing functions, and that you get 1:1 between Lua and the screen.
One problem with this approach is that the names on the screen contain special characters. I have solved that by also providing string constants for the special characters in #1169.
In #1154, I have provided functions to get the name from an index, get the index from a name, and get a value from the index.
So if you are interested, I can use the same approach to create the following functions for sources:
- `getSourceIndex(name)` is similar to `getFieldInfo(name).id`, except that we are using the name that you see on the screen (with special chars) instead of the names in [this table](https://downloads.open-tx.org/2.3/release/firmware/lua_fields_x12s.txt).
- `getSourceName(index)` will get you the name that you see on the screen, and I believe that it would work with index values from the above function as well as `getFieldInfo(name).id`, as these _should_ be the same.
- I _think_ that `getValue(index)` would work with all of the indices, so _probably_ no need to change anything here.

Please let me know what you think about that.

offer-shmuely, 2021-12-24:
I will love to get the `getSourceName(index) `
then I can continue with my small project

jfrickmann, 2021-12-24:
After our discussion on Discord, I was planning to modify `getFieldInfo` so it can also search for an id like `getValue`. Is that still OK?
Since it is Xmas now, it will probably take a few days before I have time to do it.

offer-shmuely, 2021-12-24:
yep, that will be great as well!

jfrickmann, 2021-12-27:
Please see #1300

jfrickmann, 2021-12-27:
OK, GitHub is working again, and all checks have passed. So there is FW to test!

closed 2022-01-03 by pfeerick