EdgeTX/edgetx #1287: Mixer line names get not converted

lshems, 2021-12-22:
[XD9+.zip](https://github.com/EdgeTX/edgetx/files/7765482/XD9%2B.zip)
2.6.0 RC1.0

when loading this model in companion and starting the simu, some mixer line names will not be available in the simu mixer lines.

Check mixer 31, line 3 for instance. In companion it shows "Roll" as name for the mixer line.
When starting the radio simulator, the name for mixer line 31,3 is [gone.]()

raphaelcoeffic, 2021-12-23:
@lshems which Companion build did you use?

lshems, 2021-12-23:
As noted 2.6.0 rc1.0 from the action artefacts on window companion.

lshems, 2021-12-23:
https://github.com/EdgeTX/edgetx/actions/runs/1601646319 on arm64 update

raphaelcoeffic, 2021-12-26:
@lshems is this correct?
[image]

lshems, 2021-12-26:
This is indeed exact the issue

elecpower, 2021-12-26:
@raphaelcoeffic I vote for moving forward and focus on yaml

lshems, 2021-12-26:
I think with a yaml interpreter in lua, we could ditch the standard GUI if there would be a reload yaml file function. 

Everybody could create his favorit layout and behaviour just using LUA!!!

raphaelcoeffic, 2021-12-26:
@lshems not sure it would be so efficient. That being said, it would also require quite some stuff to offer all the capabilities of the C++ code. Also, it’s a bit off-topic for this issue 😜

lshems, 2021-12-26:
Would a reload function in lua be possible? Then I assure you it can be in done (quite easily).

closed 2022-01-02 by raphaelcoeffic