EdgeTX/edgetx #1290: Long delay when using switch as input

eshifri, 2021-12-23:
**Describe the bug**
It was reported on Discord: https://discord.com/channels/839849772864503828/839849772864503832/923210849487847455
When a 3 pos. switch is used as an input source there is an issue:
TWin says that there is no way to get the full range from the switch (-100; 0; 100) on RMTX16.
I have tried it on NV14 and I can get all three values, but there is a huge (~2s) delay, when going through zero in both directions.

**To Reproduce**
Create a new input and use 3-pos switch as source.
Change switch position from UP to -.
Observe that on the curve the point jumps from bottom left to the center instantaneously.
Change switch from - to DOWN.
Observe the point is stuck on zero for a couple of seconds.
The same in the opposite direction.
One can create the mixer using this input and observe the same behavior on the channel monitor.

**Expected behavior**
There should be no delay.

**General information**
 - EdgeTX version 2.6RC1.
 - Hardware NV14 and TX16S.

raphaelcoeffic, 2021-12-23:
Checked on `main` with simulator: no issue. Now moving on to hardware.

eshifri, 2021-12-23:
I have tried it on NV14 - TWin initially observed it on TX16S.
Some additional tests show that it is switch-dependant: SF works fine; SB shows delay in the central position; SG does not work at all (on NV14)
[model1.zip](https://github.com/EdgeTX/edgetx/files/7770604/model1.zip)
Model file attached. Please take a look at the input16.

eshifri, 2021-12-23:
Interesting: I cannot reproduce it on TX16S either. With any switch.
Let me rebuild the NV14 code.

raphaelcoeffic, 2021-12-26:
@eshifri could you please add `radio.yml` as well?

eshifri, 2021-12-26:
[radio.zip](https://github.com/EdgeTX/edgetx/files/7777262/radio.zip)
here it is.

eshifri, 2021-12-28:
Turned out the delay was caused by "screenshot" function assigned to SWB-

closed 2021-12-28 by eshifri

pfeerick, 2021-12-29:
Yeah, I've forgotten that one a few times... I use SWC as my screenshot one on NV14, and SH on TX16S (i.e. momentary ones)

eshifri, 2021-12-29:
On TX16S it is not a problem: one can use T5 UP and DOWN.
On NV14 I use SC in arm/disarm sequence, so I assigned screenshot to SWB- and naturally forgot about it.
And unfortunately there is no way to disable this GF. :-) Maybe new feature request? :-)

pfeerick, 2021-12-29:
If you don't use GFs generally, you could disable them across the whole model, but yeah, an enable toggle would be nice for that, for when you don't actually want it, but like to leave it setup.