EdgeTX/edgetx #1292: special function to select screens (main and telemetry screens)

ChrisOhara57, 2021-12-23:
Hello

I do use a radio with black and white screen (x9lite S).

I have up to two telemetry screens (nums) and one or two telemetry scripts for one model.
While flying it is difficult to switch between different screens (or select the desired screen).
Offen I do need more than one attempt because I'm pushing long instead of short or vice versa.
(Sometime I have only forgotten the right sequence. LOL )
It drags my attention away from my plane.

It would be great to have a Special Function that does switch between the screens (main screen and telemetry screen 1, 2, 3, 4)
This would enable to:
     select by logical switches or normal switches
     (and therefore: by flightmodes, telemetry values, . . . . . . , or defining one of the screens as "start screen")
Even a script that is running in background could then draw its attention by calling the special function "or alter a logical switch" and activating its screen.

Having no experience with colored displays I can not say whether this would make sense for widgets too.

regards Chris

ChrisOhara57, 2021-12-27:
Hi

I did a little more thinking about the special function.

Minimum would be:
    SwitchScreen  [X]    (X being: 0 for main, 1 to 4 for telemetry screen)

"Golden edition" would be:
    SwitchScreen  [X]  [Y] (Y beeing: [-] stay with new screen; Y=1,2,3,4,5..... return to old screen after Y seconds)
  also with Lua-functions
    X=lcd.getScreen()  returns the Number of the active screen
    lcd.setScreen(X)   sets the screen X to be the active on 

I also did think of a small example to demonstrate how this functionality could be useful.
In this demo it could optimize a virtual-10-postion-switch.
(I do a lot of different announcement of telemetry values, and already run out of switches.)

When I turn the rotary encoder a global variable is increase or decrease.
This global variable is used to set logical switches ("position switch").
Now when I push a defined momentary switch the telemetry value referring to the value of the global variable is announced.

The drawback of the solution is:
I need to activate the telemtry screen every time I want to switch to another value.
And I have to select the original screen manually.

With the SwitchScreen function operations could be reduced:
  When I push the momentary switch 
    - the value for the active position is announced
    - additionally the screen for the virtual-10-position-switch is displayed
    - I could change the switch to another position if I want
    - after a few seconds the display would set to the original screen again (in my case the INAV script)
So I would only need one momentary switch for selecting and playing a high number of different telemetry values with a minimum of manual interactions. 

regards Chris

https://user-images.githubusercontent.com/80360397/147465436-3c86eff9-71c4-42da-8821-756b738c1b8b.mp4

[image]
[image]
[vSwtch.txt](https://github.com/EdgeTX/edgetx/files/7779864/vSwtch.txt)

jfrickmann, 2021-12-30:
I believe that the reason why navigation buttons are not available to Lua, is that we do not want a misbehaving script to hijack the radio's GUI. I see the same risk with this idea.
Instead I suggest that you make one Ltelemetry ua script that can show either of your menus. This is easy to do, and with Lua's facilities to dynamically load code, you can easily implement it all in Lua.

ChrisOhara57, 2021-12-30:
By hijack you mean something like an endless loop. Two scripts that take away the screen in turns.
Yes I see that that could be possible.

ChrisOhara57, 2021-12-31:
If that happens a hardware switch would be needed to turn that function off.
Or turn the receiver off and remove the SD-card.
I think that would happen seldom but when it does it would cause real panic.
So think I will close the issue.
Thanks for the comment
Regards Chris

closed 2021-12-31 by ChrisOhara57

jfrickmann, 2021-12-31:
Here is a Lua telemetry script that should accomplish what you want to do. I have not tested it, so there may be typos or other goofs, but hopefully you get the idea!
```
-- Telemetry script to run two telemetry scripts on the same screen
-- Author: Jesper Frickmann

local script1 = "/SCRIPTS/TELEMETRY/script1.lua"
local script2 = "/SCRIPTS/TELEMETRY/script2.lua"
local switch = getFieldInfo("sa").id
local error1
local error2

local function init()
  local chunk
  
  chunk, error1 = loadScript(script1)
  if chunk then
    script1 = chunk() -- Saving memory by re-using script1 var
    if script1.init then
      script1.init()
      script1.init = nil -- Not needed anymore, save memory
    end
  end
  
  chunk, error2 = loadScript(script2)
  if chunk then
    script2 = chunk()
    if script2.init then
      script2.init()
      script2.init = nil
    end
  end
end

local function background()
  if not error1 and script1.background then
    script1.background()
  end

  if not error2 and script2.background then
    script2.background()
  end
end

local function run(event)
  if getValue(switch) > 0 then
    if error1 then
      lcd.drawText(0, 10, error1, SMLSIZE)
    else
      script1.run(event)
    end
  else
    if error2 then
      lcd.drawText(0, 10, error2, SMLSIZE)
    else
      script2.run(event)
    end
  end
end

return { init = init, background = background, run = run }
```

ChrisOhara57, 2021-12-31:
I think I got it. Will try (but not the next days, need to think of a project for that)

In init and background the two scripts are address in sequence.
While in run function the call depends on the position of the switch.
Using a logical switch or a global variable instead of a hardware switch would enable to do a free selection (between the two scripts).
It would even be possible to generate an endless loop ( by calling each other),
But this wouldn't be dramatic cause switching to another telemetry screen would still be possible.
Interesting , Thanks a lot
Regards Chris

PS: I've read you live in North Carolina. I hope to do some flying in South Carolina in February and March (if our flights don't get canceled due to Corona)

jfrickmann, 2022-01-04:
Do you fly sailplanes? If you do, then please come join us some time!

ChrisOhara57, 2022-01-04:
Thanks for the invitation. I would love to try.
But I doubt that I will have the opportunity (Time is reserved for family).
And distances in US aren't that short (compared to Germany).
It would still be a few hours to travel I think.