EdgeTX/edgetx #1295: Issue after "clear" curves

Tiefflieger68, 2021-12-26:
There is an issue at the curves.
Tested with EdgeTX 2.5 and 2.6 Nighties. Everything is fine at openTX 2.3.14.
Same result with Companion Simulator or Radio (both TX16s)

Test:
modify Curve 1 - fine to this point. You can edit Curve 1 again without issue
"clear" Curve 1 - fine to this point. 
modify Curve 1 again
now Curve 2 or 3 appears by itself with random values - ISSUE !

jfrickmann, 2021-12-27:
You referred this issue from the Lua thread on Discord, is that because it is caused by a Lua script using `setCurve`? 
If so, please give us a test script so we can look into it!

Tiefflieger68, 2021-12-27:
no - nothing to do with LUA. This was my first thougt, cause I was thumbeling about this issue when I tested my LUA script.

To test, just go in Model Menue (hard TX16s or Companion simulator) and follow flow below
modify Curve 1
"clear" Curve 1
modify Curve 1 again
now Curve 2 or 3 appears by itself with random values (sometimes 1 or 2 more tries needed)

Tiefflieger68, 2021-12-27:
I've tested this with 2 different TX16s (edgeTX 2.5 and 2.6) and Companion Simulator 2.5.
Always with the same result

jfrickmann, 2021-12-27:
I have now played a little with Companion 2.5 and 2.7 for Windows (#1186) and my own Linux build from a branch off main. And so far, I have not been able to reproduce the problem.

pfeerick, 2021-12-28:
It seems like the requirements may be a little more specific - I can readily repeat this with 2.5 simulator when at the edit stage I choose custom rather than standard curve, then clear the curve, wait for settings to be saved, and then reload the radio data, and viola, unwanted curve 2 is present. 

https://user-images.githubusercontent.com/5500713/147514576-a3722dd2-e66a-42ae-ab32-6c4d52d468d1.mp4

Tiefflieger68, 2021-12-28:
hm - sometimes not that easy to reproduse the issue...
(I'm not able to record the Companion Simulator with the Win10 Gaming Recorder)
It seems the issue appears if you have programmed some functions. Find attached otx file.
Select "TX16s" in Profile and load the contained model.
Clear Curve 1 and 2. Then Curve 2 will appear again with random values.
[curveissue.zip](https://github.com/EdgeTX/edgetx/files/7784404/curveissue.zip)

PR #1339 (2022-01-04): Fixed clearing curves
Resolves #1295

Summary of changes:
- "Clear curve" now correctly handles shifts in global point array.
- Change applied to B&W targets as well.
commit 44cdb449 (2022-01-04): Fixed clearing curves

Resolves #1295
commit 3ab603f7 (2022-01-04): Fixed clearing curves

Resolves #1295
closed 2022-01-04 by pfeerick