EdgeTX/edgetx #1296: ISRM  shows LR12 instead of ACCST/D16 and works without telemetry even if telem. is set on

wimalopaan, 2021-12-26:
I have a X9d+2019 with ISRM-module (with fw 2.1.6).

With OpenTx all is fine in ACCESS or in ACCST/D16 mode. Telemetry for ACCST works well.

With EdgeTx I get the options ISRM and LR12. Binding to an RX8R with ACCST using LR12 works, it asks aboiut telemetry but then no sensors show up.

raphaelcoeffic, 2021-12-26:
Where do you see LR12? I can see only `ACCESS` and `ACCST D16`:
[image]
[image]

Could you please post your `radio.yml` file? Did you compile it yourself, or is it a GH build?

wimalopaan, 2021-12-26:
```
version: 221
calib: 
   Rud:
      mid: 1000
      spanNeg: 790
      spanPos: 807
   Ele:
      mid: 1021
      spanNeg: 763
      spanPos: 791
   Thr:
      mid: 972
      spanNeg: 781
      spanPos: 777
   Ail:
      mid: 1085
      spanNeg: 837
      spanPos: 754
   POT1:
      mid: 1061
      spanNeg: 1044
      spanPos: 970
   POT2:
      mid: 1037
      spanNeg: 1020
      spanPos: 993
   SLIDER1:
      mid: 1016
      spanNeg: 918
      spanPos: 892
   SLIDER2:
      mid: 1013
      spanNeg: 946
      spanPos: 899
currModel: 0
contrast: 25
vBatWarn: 65
txVoltageCalibration: 0
backlightMode: backlight_mode_all
antennaMode: MODE_PER_MODEL
disableRtcWarning: 0
keysBacklight: 0
internalModule: TYPE_ISRM_PXX2
trainer: 
   mix: 
      0:
         srcChn: 0
         mode: REPL
         studWeight: 100
      1:
         srcChn: 1
         mode: REPL
         studWeight: 100
      2:
         srcChn: 2
         mode: REPL
         studWeight: 100
      3:
         srcChn: 3
         mode: REPL
         studWeight: 100
view: 0
fai: 0
beepMode: mode_nokeys
alarmsFlash: 0
disableMemoryWarning: 0
disableAlarmWarning: 0
stickMode: 0
timezone: 0
adjustRTC: 0
inactivityTimer: 10
telemetryBaudrate: 0
splashMode: 0
hapticMode: mode_nokeys
switchesDelay: 0
lightAutoOff: 2
templateSetup: 0
PPM_Multiplier: 0
hapticLength: 2
beepLength: 2
hapticStrength: 2
gpsFormat: 0
unexpectedShutdown: 0
speakerPitch: 0
speakerVolume: 12
vBatMin: 60
vBatMax: 80
backlightBright: 0
globalTimer: 3549
bluetoothBaudrate: 0
bluetoothMode: TELEMETRY
countryCode: 0
pwrOnSpeed: 0
pwrOffSpeed: 0
imperial: 0
jitterFilter: 0
disableRssiPoweroffAlarm: 0
USBMode: 0
jackMode: 0
ttsLanguage: "en"
beepVolume: 2
wavVolume: 4
varioVolume: 2
backgroundVolume: 3
varioPitch: 0
varioRange: 0
varioRepeat: 0
auxSerialMode: MODE_NONE
slidersConfig: 
   SLIDER1:
      type: with_detent
      name: 
   SLIDER2:
      type: with_detent
      name: 
potsConfig: 
   POT1:
      type: with_detent
      name: 
   POT2:
      type: with_detent
      name: 
backlightColor: 0
switchConfig: 
   SA:
      type: 3pos
      name: 
   SB:
      type: 3pos
      name: 
   SC:
      type: 3pos
      name: 
   SD:
      type: 3pos
      name: 
   SE:
      type: 3pos
      name: 
   SF:
      type: 2pos
      name: 
   SG:
      type: 3pos
      name: 
   SH:
      type: toggle
      name: 
   SI:
      type: toggle
      name: 
bluetoothName: ""
ownerRegistrationID: "\x16\x11829294"
uartSampleMode: 0

```

wimalopaan, 2021-12-26:
[image]

wimalopaan, 2021-12-26:
[image]
[image]

raphaelcoeffic, 2021-12-26:
Ok, so self-compiled + translation error.

wimalopaan, 2021-12-26:
What about telemetry? Should be there?

raphaelcoeffic, 2021-12-26:
@pfeerick can you please give it a try on your shiny X9D+2019?

raphaelcoeffic, 2021-12-26:
Two issues here:
- defining `INTERNAL_MODULES=PXX1,PXX2` instead of `INTERNAL_MODULES=PXX1;PXX2` basically prevents definition from the proper compilation variables.
- defining `INTERNAL_MODULE_PXX1` on a `X9D+2019` leads to a translation f..-up because of this beauty:
```
#if defined(INTERNAL_MODULE_PXX1) || defined(PCBHORUS)
#define LEN_ISRM_RF_PROTOCOLS          "\006"
#define TR_ISRM_RF_PROTOCOLS           "ACCESS""D16\0  ""LR12"
#else
#define LEN_ISRM_RF_PROTOCOLS          "\012"
#define TR_ISRM_RF_PROTOCOLS           "OFF\0      ""ACCESS\0   ""ACCST D16\0""ACCST LR12""ACCST D8"
#endif
```

However, fixing the last one is a bit tricky ATM and needs more investigations.

wimalopaan, 2021-12-26:
Was using wrong compilation flags: only select PXX2: -DINTERNAL_MODULES="PXX2" 

Don't -DINTERNAL_MODULES="PXX1;PXX2"

closed 2021-12-26 by wimalopaan