EdgeTX/edgetx #1297: Can't update internal ISRM (but with OpenTx)

wimalopaan, 2021-12-26:
... after "device reset" follows "device not responding".

Info: with OpenTx this works well.

raphaelcoeffic, 2021-12-26:
@wimalopaan version info?

wollenagold, 2021-12-26:
Hi guys
I made an report in EdgeTS/SDCard with the following Question, it seems this is perhaps a same problem:
Changing on to EdgeTX i don't find my internal Access-Module when i'm opening the internal RF-selection, only XJT is displayed or OFF.
Is it possible that i put a wrong Version of 2.5.0 Dauntless? Functionally has been ok, only the RF-selection has an issue.
FrSky X12S, switching from OpenTX 2.3.14, RF-Modul = ISRM normally switchable from Access to ACCST, I have five ACCST and 5 ACCESS-Reveivers, all are working under OpenTX
Today, one day later new attempt:
Changing back to OpenTX 2.3.14 with changing the Models- and Radio-Directorys from the back-up will give me back the complete functionally of the ISRM-Modul inside the TX under OTX. The HF-Links to the receivers are all ok.

Thank you for Help which make me able to can try it again with Edge.
Wolfgang

raphaelcoeffic, 2021-12-26:
@wollenagold Please note that this has nothing to do with the present issue. If you have questions on how to use the proper firmware, please come to our Discord server. Github issues are for software bugs, not questions.
 
> Changing on to EdgeTX i don't find my internal Access-Module when i'm opening the internal RF-selection, only XJT is displayed or OFF.

This is only available from 2.6.0 on (RC1 or current nightly). On that version, you need to go to "Hardware settings" to select the internal module installed. The model settings can only turn the module selected in the hardware settings ON/OFF, not select another.

> Is it possible that i put a wrong Version of 2.5.0 Dauntless? Functionally has been ok, only the RF-selection has an issue.

You will need 2.6.0 for that.

raphaelcoeffic, 2021-12-26:
@pfeerick can you please try this out and maybe make a trace with your shiny new red toy in case it does not work?

wimalopaan, 2021-12-26:
> @wimalopaan radio / version info

This was related to #1296.

closed 2021-12-26 by wimalopaan

wollenagold, 2021-12-26:
> Hi guys I made an report in EdgeTS/SDCard with the following Question, it seems this is perhaps a same problem: Changing on to EdgeTX i don't find my internal Access-Module when i'm opening the internal RF-selection, only XJT is displayed or OFF. Is it possible that i put a wrong Version of 2.5.0 Dauntless? Functionally has been ok, only the RF-selection has an issue. FrSky X12S, switching from OpenTX 2.3.14, RF-Modul = ISRM normally switchable from Access to ACCST, I have five ACCST and 5 ACCESS-Reveivers, all are working under OpenTX Today, one day later new attempt: Changing back to OpenTX 2.3.14 with changing the Models- and Radio-Directorys from the back-up will give me back the complete functionally of the IRSM-Modul inside the TX under OTX. The HF-Links to the receivers are all ok.
> 
> Thank you for Help which make me able to can try it again with Edge. Wolfgang

Thank for the Information. That helps me so my comment is closed.
Quick help feels good
Wollenagold