EdgeTX/edgetx #1298: Global widget setup

Mantamatt1, 2021-12-26:
Tx16s
Fw latest.
I've mentioned global widgets before and was told it's not going to be done but I thought I would give it another shot now Joshua bardwell has now mentioned it in one of his videos.
I would like a way to set up my home screen with all the widgets I use etc as most the time I'm using the same widgets and laid out how I like them. So when I create a new model all my widgets are already in place. I know I can create a template but I find that more clunky and more of a work around as I have to find the template, then copy, remember to change the name ect! Or with global widgets I can just create new model? Would find that a lot more premium and so would bardwell lol

raphaelcoeffic, 2021-12-26:
This is close to the idea of templates.

pfeerick, 2021-12-27:
Dulicate of #351

closed 2021-12-27 by pfeerick