EdgeTX/edgetx #1299: Changing radio settings hardware internal module does update each existing model internal rf mode

elecpower, 2021-12-26:
**Describe the bug**
Radio settings changing hardware internal module does reset / update each existing model internal rf mode or warn they are incompatible.

**To Reproduce**
Steps using RM TX16S sim
Edit Internal module in hardware settings say from MULTI to CRSF and exit
Open an existing model setup
Go to Internal Rf and mode is for previous internal module along with all the supporting settings.
Click on the mode value and the drop down selection is only those compatible with new internal module.

**Expected behaviour**
At the time of changing the hardware internal module reset every model internal rf mode to OFF or warn that the model settings are incompatible.

**Screenshots**
[image]

**General information**
 - EdgeTX version main branch
 - Hardware simulated RM TX16S

raphaelcoeffic, 2021-12-26:
Yeah, it was never really intended to update all the models. It might be a bit tricky as well, as it might require additional settings which would not be available anyway. Let’s call this a known limitation.

elecpower, 2021-12-26:
The safe approach would be to set all internal Rf modes to OFF then no need to worry about mode specific settings.

raphaelcoeffic, 2021-12-26:
> The safe approach would be to set all internal Rf modes to OFF then no need to worry about mode specific settings.

You mean when a model with an incompatible module type set is loaded? That should be doable easily, indeed. And normally, when the module is set to OFF, you notice soon enough ;-)

elecpower, 2021-12-26:
Better for a model not to operate than to operate unexpectedly or potentially dangerously

pfeerick, 2021-12-27:
Ok, so at time of model load, if the internal RF module != configured hardware model, internal RF == OFF.

elecpower, 2021-12-27:
Also YAML files and the ability to manipulate outside official code is going to create opportunities for data integrity issues. So this critical test needs to occur on model load.

pfeerick, 2021-12-27:
After actually looking though this further, this only applies to the *active* model - there is already some safeguard that sets the type to MODULE_TYPE_NONE if `g_eeGeneral.internalModule` doesn't match - probably `isInternalModuleAvailable()` - which kicks in as soon as you change models, or power cycle... so the only change that is needed is for the internal RF to be turned off when the module setting is changed in the hardware settings...

PR #1316 (2022-01-03): Clear internal RF module settings if radio level setting changed
Resolves #1299 

Ensures if you change the internal module at the radio level, that the currently active model internal rf settings are cleared, thus inhibited. This is not needed on model load, as there is a model load check which inhibits the dissimilar settings being made active. 

Summary of changes:
- On colorlcd, if you chose an option for the internal protocol other than the current one, the active model level setting is cleared  - so you can view the list and abort or re-select the current internal module without loosing  [... cut]
closed 2022-01-03 by pfeerick