EdgeTX/edgetx #1301: Weird interaction of SWF and SF

eshifri, 2021-12-27:
**Describe the bug**
Special Functions have weird reaction f=to switch F.

**To Reproduce**
1. Create a SF1 "Play Track".
2. Select SF- as a source.
3. Observe that the track is not played when SWF is set to middle position.
4. Create another "Play Track" SF (SF2).
5. Select SWF_DOWN as a source.
6. Set SWF to "-".
7. Observed the second SF (SF2) being highlighted and played. SF1 is never triggered.

HW debug page shows proper changes for SWF position.
One can create LSs based on SWF and they work properly. (So the work-around for the described problem is to create intermediate logical switches instead of using SWF directly.)

The problem was introduced recently - everything was working fine ~ 2 weeks ago.

**General information**
 - EdgeTX current
 - Hardware NV14

pfeerick, 2021-12-28:
It's even wierder than that... I initially created a special function on a build from yesterday with SWF- as the condition, and it worked perfectly... But as soon as I added SWF_down it acts like SWF_down is down, and SWF- and SWF_up are the same... i.e it is behaving more like it is a two pos switch. And as you say, hardware page and logical switch a=x can detect SWF moving properly. 

The three positions of SWG are fine though.

eshifri, 2021-12-28:
Looks like it also depends on the SF selected: e.g. for Haptic everything works correctly.

PR #1313 (2021-12-29): Add switches definitions for NV14 #1301
Fixes #1301 

Summary of changes: add definitions for NV14 switches for CF support
closed 2021-12-29 by pfeerick