EdgeTX/edgetx #1306: Total mess with lua-mixer scripts

wimalopaan, 2021-12-28:
I have this simple useless lua script:

```
local input = {
    {"Input1", SOURCE},
    {"Input2", SOURCE},
    {"Input3", SOURCE},
    {"Input4", SOURCE},
    {"V1", VALUE, 0, 100, 10},
    {"V2", VALUE, 0, 100, 2}
 };
    
local output = { 
    "Incremental", 
    "I2", 
    "I3", 
    "I4", 
    "I5", 
    "I6", 
};
    
local function run(s1, s2, s3, s4, v1, v2)
    return s1, (s1 + s2), (s1 + s2 + s3), (s1 + s2 + s3 + s4), v1, v2;
end
    
return {
    input = input,
    output = output,
    run = run
};

```
To reproduce the bug:

1) insert this script as LUA2 (not LUA1)
2) you will not see the inputs and outs on that page
3) go to the insert dialog of LUA1: there you see the inputs and outputs of the script in LUA2.

The model yml:

```
header: 
   name: "MODELL37"
   bitmap: ""
telemetryProtocol: 0
thrTrim: 0
noGlobalFunctions: 0
displayTrims: 0
ignoreSensorIds: 0
trimInc: 0
disableThrottleWarning: 0
displayChecklist: 0
extendedLimits: 0
extendedTrims: 0
throttleReversed: 0
beepANACenter: 0
mixData: 
 -
   weight: 100
   destCh: 0
   srcRaw: I0
   carryTrim: 0
   mixWarn: 0
   mltpx: ADD
   offset: 0
   swtch: NONE
   flightModes: 000000000
   delayUp: 0
   delayDown: 0
   speedUp: 0
   speedDown: 0
   name: ""
 -
   weight: 100
   destCh: 1
   srcRaw: I1
   carryTrim: 0
   mixWarn: 0
   mltpx: ADD
   offset: 0
   swtch: NONE
   flightModes: 000000000
   delayUp: 0
   delayDown: 0
   speedUp: 0
   speedDown: 0
   name: ""
 -
   weight: 100
   destCh: 2
   srcRaw: I2
   carryTrim: 0
   mixWarn: 0
   mltpx: ADD
   offset: 0
   swtch: NONE
   flightModes: 000000000
   delayUp: 0
   delayDown: 0
   speedUp: 0
   speedDown: 0
   name: ""
 -
   weight: 100
   destCh: 3
   srcRaw: I3
   carryTrim: 0
   mixWarn: 0
   mltpx: ADD
   offset: 0
   swtch: NONE
   flightModes: 000000000
   delayUp: 0
   delayDown: 0
   speedUp: 0
   speedDown: 0
   name: ""
expoData: 
 -
   mode: 3
   scale: 0
   srcRaw: Ail
   carryTrim: 0
   chn: 0
   swtch: NONE
   flightModes: 000000000
   weight: 100
   name: ""
   offset: 0
   curve: 
      type: 1
      value: 0
 -
   mode: 3
   scale: 0
   srcRaw: Ele
   carryTrim: 0
   chn: 1
   swtch: NONE
   flightModes: 000000000
   weight: 100
   name: ""
   offset: 0
   curve: 
      type: 1
      value: 0
 -
   mode: 3
   scale: 0
   srcRaw: Thr
   carryTrim: 0
   chn: 2
   swtch: NONE
   flightModes: 000000000
   weight: 100
   name: ""
   offset: 0
   curve: 
      type: 1
      value: 0
 -
   mode: 3
   scale: 0
   srcRaw: Rud
   carryTrim: 0
   chn: 3
   swtch: NONE
   flightModes: 000000000
   weight: 100
   name: ""
   offset: 0
   curve: 
      type: 1
      value: 0
thrTraceSrc: Thr
switchWarningState: AuBuCuDuEuFuGuHu
rssiSource: none
thrTrimSw: 0
potsWarnMode: WARN_OFF
scriptsData: 
   1:
      file: "test"
      name: ""
inputNames: 
   0:
      val: "Que"
   1:
      val: "H\x99h"
   2:
      val: "Gas"
   3:
      val: "Sei"
potsWarnEnabled: 0
screenData: 
   0:
      LayoutId: "Layout2P1"
      layoutData: 
         options: 
            0:
               type: Bool
               value: 
                  boolValue: 1
            1:
               type: Bool
               value: 
                  boolValue: 1
            2:
               type: Bool
               value: 
                  boolValue: 1
            3:
               type: Bool
               value: 
                  boolValue: 1
            4:
               type: Bool
               value: 
                  boolValue: 0
view: 0
modelRegistrationID: "WM"

```

wimalopaan, 2021-12-28:
[image]
[image]
[image]

wimalopaan, 2021-12-28:
Looks as if there is an off-by-one error in some internal loop.

pfeerick, 2021-12-28:
Is this new behaviour? Would you be able to try https://github.com/EdgeTX/edgetx/commit/e4369af0480e9e7afd215450e3592030045189d5 and see if it still breaks the same way? Just so can verify it isn't a side effect of the change to `LUA=YES => LUA_MIXER`

wimalopaan, 2021-12-28:
mmh, the link you mentioned maybe wrong? Is just a color change ...

pfeerick, 2021-12-28:
That is the commit immediately prior to LUA_MIXER being added ;)

wimalopaan, 2021-12-28:
Ah ok. Just give me a hand: tried to get this via zip, but that does not include the submodules as libopenui.

wimalopaan, 2021-12-28:
It is excactly the same issue with that commit.

wimalopaan, 2021-12-28:
Tried the same commit with another cmake

`cmake -DUSB_SERIAL=Off -DTRANSLATIONS=EN -DPCB=X10 -DPCBREV=TX16S -DLUA=YES -DLUA_MIXER=YES -DGVARS=YES -DINTERNAL_MODULES="MULTI"  ..`

to exclude translation-issues or some other non-standard effects.
But the problem remains.

jfrickmann, 2021-12-28:
It was a dumb bug - it should be fixed now in #1312

closed 2021-12-30 by pfeerick