EdgeTX/edgetx #1307: FlySky Nirvana NV14 freezes/crashes when disconnected from USB 5v charger

ssALviK, 2021-12-28:
FlySky Nirvana NV14 crashes / freezes and goes into emergency mode if usb charging cable is disconnected while charging from 5V power brick. 

This issue happens if there is a external RF module connected in the JrBay ("TBS Crossfire TX Lite" in my case).

FW: EdgeTX 2.7.0 (a1e03ed7) merge/nightly (happens on 2.6.0 RC1 cdeb92a too)

[image]
[image]
[image]

pfeerick, 2021-12-29:
Have you tried *not* putting it into `USB Storage (SD)` mode (since a charger is not a computer :grin:)? Press the left joystick hat for exit when the menu is shown, it'll stop popping up then. It shouldn't be crashing anyway, but it would be interesting to see if it is specifically related to the reload that happens when the USB is disconnected when in storage mode.

ssALviK, 2021-12-29:
Tomorrow morning i'll test if it does crash in other instances too. 
I'm not sure if it charges or not if you don't pick any usb option. 
Either way tomorrow i'll do some more tests. 
(btw it crashes even when connected to pc and jrbay is populated)

pfeerick, 2021-12-29:
Thanks. The radio *can* enable/disable charging, but I am under the impression it is enabled.

ssALviK, 2021-12-29:
Hello!
So, I've been testing various scenarios and it pretty much crashes/freezes and goes into Emergency Mode most of the times when you connect and disconnect usb cable regardless of what USB mode you choose or not. 
It does this only when JRBAY is populated. 
No issues when JrBay is empty. 
(Also, I've tried a diffrent pc and my phone, the issues remain but when i connect it to the phone via OTG cable my phone crashes LMAO)

pfeerick, 2021-12-30:
Will have to investigate this further, and see how to best reproduce it. With a HappyModel ES24TX module in the JR bay, enabled, and even with the ELRSv2 Lua running at the time of connecting USB, I was unable to reproduce a custom build of current `main`. Disconnecting the USB when either plugged into computer or USB powerbank did not result in EM or reset/restart. 😢

ssALviK, 2022-01-01:
IDK, maybe it has to do something with NV14 & TBS Crossfire TX Lite combo?
BTW, I've tried 2.6.0 RC1 too, it also results in EM crash/freeze. 
Also, my NV14 hasn't had any HW modification or anything done to it. Everything is in stock form (HW wise). 
TBS Crossfire is updated to latest FW.

pfeerick, 2022-01-01:
In that case, it *could* be a hardware problem, as there is a power mod around to fix some issues with module power... I'm just checking in with another user to see if that could be relevant.

eshifri, 2022-01-05:
I got similar problem in 2.6RC2 without external module and when connecting NV14 to the computer.
If NV14 is connected by USB and is ON disconnecting the cable causes crash (in any USB mode).
No HW mods here.

yds, 2022-01-05:
FWIW, tried to reproduce this issue with both Crossfire and ExpressLRS TXen on a stock unmodded Nirvana and I couldn't get it to crash regardless of USB mode nor if it was a PC or charger only.. dunno what to say, but no crashes nor emergency mode for me..

eshifri, 2022-01-05:
Was crashing for me all the time yesterday, but I cannot reproduce it today (same SW version).

gagarinlg, 2022-01-05:
Maybe it depends on the charge of the batteries?

eshifri, 2022-01-05:
My batteries were close to fully charged. I made an attempt to update the SW with "buddy" yesterday, but did not go through with it before doing all these tests. Performed multiple reboots after this, so I doubt this is relevant though. The system was reliably crashing every time I disconnected the USB cable. Today it is working as reliably as it crashed yesterday.

chipset2011, 2022-01-07:
This problem is related to poor power supply inside the NV-14, probably you have TBS crossfire lite turned on at high power, such problems are observed at power above 100 mv. I was able to solve this problem by changing the power subsystem of the nv-14 itself and the power supply of the jr compartment.

OG-ALviK, 2022-01-07:
Nope, my TBS lite was set to 10-25mw.

stefan73, 2022-01-16:
Cannot reproduce this with my Nirvana which has the power-mod (makes power more stable).

Here is my suspicion on why there are different test results:
If you toggle to SD Mode, the external module is powered off. There are old HW revisions, where the module does NOT power off (can be changed by the published HW mod for there radios) ever. The bay is for these old HW revisions always on.
@eshifri Does your module power down when you select SD mode?

eshifri, 2022-01-16:
As I have mentioned, I did not have external module installed. My TX is old and, most likely, has the original HW revision.
The system was repeatedly crashing one evening and is working correctly since then.

OG-ALviK, 2022-01-16:
My module powers off completely when the radio is off. HW revision is NV14-V1.6, year; 2018.07.26
No HW mods done to it. 
Even tho i'm very tempted to do it.

closed 2022-11-17 by OG-ALviK