EdgeTX/edgetx #1310: In the character set at EDGE the special characters are missing

koaxheli, 2021-12-28:
In the character set at EDGE the special characters are missing
Decimal from 147 to 158.
In OTX they are present see screenshot
[image]
.

pfeerick, 2021-12-29:
Since this seems to be a Lua script, can we get the source for that as well so that the fault can be reproduced locally? ;)

koaxheli, 2021-12-29:
Here is the Lua to the missing special characters.
With Edge 2.6 Nightly
Path: \SCRIPTS\TOOLS\WizardTest.lua
[image]
[image]
[WizardTest.zip](https://github.com/EdgeTX/edgetx/files/7788210/WizardTest.zip)

pfeerick, 2021-12-29:
Excellent! Thanks for that! :) @richardclli Is there any chance this is related to translation handling in some way? I've verified this on TX16S on `main` and also against v2.5.0, so it's not a new breakage.

koaxheli, 2021-12-29:
The shift of two digits is not the problem, but from 149 on there are no more characters.
The special characters like ä, ö and ü are missing.

pfeerick, 2021-12-29:
I didn't say it is **the** problem... ;)

>*There is also*

koaxheli, 2021-12-31:
Hello,
And where can I get the German translation?

All the best in the new year and a happy new year 2022

pfeerick, 2021-12-31:
Self-build with the `-DTRANSLATIONS=DE` flag. Or tell me the build options, target radio (from companion) and preferred version, and a build will magically appear here within 24 hours hours 😁

koaxheli, 2022-01-01:
The target radio is a Radiomaster TX16S, preferred version.
EdgeTX "Santa" 2.6.0 latest
The latest status is see attachment.
[image]
.
[image]

pfeerick, 2022-01-03:
Ok, maybe not 24 hours! Flex won't be needed, as that is on by default, And just a clarification - is LUA ticked because you want the lua custom mixer support, or because you want lua (as that checkbox is really for the lua mixer scripts - which isn't confusing *at all*). Either way, both are attached. I'll do another build probably tomorrow since RC2 is likely to be out then. 

`cmake -Wno-dev -DPCB=X10 -DPCBREV=TX16S -DTRANSLATIONS=DE -DBLUETOOTH=Y ..`
`cmake -Wno-dev -DPCB=X10 -DPCBREV=TX16S -DTRANSLATIONS=DE -DBLUETOOTH=Y -DLUA_MIXER=Y ..`

[build-TX16S.zip](https://github.com/EdgeTX/edgetx/files/7800878/build-TX16S.zip)

koaxheli, 2022-01-24:
Since the EdgeTX "Santa Claus" v2.6.0 is released I could have a version with Bluetooth and in German, thanks.

pfeerick, 2022-01-24:
Yes, I'll be building all the language variants today, so you will get it soon :)

richardclli, 2022-01-25:
> Excellent! Thanks for that! :) @richardclli Is there any chance this is related to translation handling in some way? I've verified this on TX16S on `main` and also against v2.5.0, so it's not a new breakage. There is also a two character shift present between OTX and ETX... i.e. the down arrow is 129 on OTX, but 131 on ETX.

ETX translations are complete different from OTX,  OTX used a fixed mapping of special characters besides En and extra (icons).  Now because of B/W uses fixed font files, a fixed mapping of special characters are used per translation.  But color lcd uses a dymanic approach, CJK and other languages used the same method that will parse thee translation files and obtain a custom set of special characters.

jfrickmann, 2022-01-25:
Please see #1169. I believe that it makes this one redundant. Please try this script:
```
local charNames = {
  "CHAR_RIGHT",
  "CHAR_LEFT",
  "CHAR_UP",
  "CHAR_DOWN",
  "CHAR_DELTA",
  "CHAR_STICK",
  "CHAR_POT",
  "CHAR_SLIDER",
  "CHAR_SWITCH",
  "CHAR_TRIM",
  "CHAR_INPUT",
  "CHAR_FUNCTION",
  "CHAR_CYC",
  "CHAR_TRAINER",
  "CHAR_CHANNEL",
  "CHAR_TELEMETRY",
  "CHAR_LUA"
}

local chars = {
  CHAR_RIGHT,
  CHAR_LEFT,
  CHAR_UP,
  CHAR_DOWN,
  CHAR_DELTA,
  CHAR_STICK,
  CHAR_POT,
  CHAR_SLIDER,
  CHAR_SWITCH,
  CHAR_TRIM,
  CHAR_INPUT,
  CHAR_FUNCTION,
  CHAR_CYC,
  CHAR_TRAINER,
  CHAR_CHANNEL,
  CHAR_TELEMETRY,
  CHAR_LUA
}

local function run(event)
  local x, y = 0, 0
  lcd.clear()
  for i = 1, #chars do
    lcd.drawText(x, y, charNames[i] .. " = " .. chars[i])
    y = y + 20
    if y > LCD_H - 20 then
      x = x + 160
      y = 0
    end
  end
  
  return 0
end

return { run = run }
```

pfeerick, 2022-01-25:
> Since the EdgeTX "Santa Claus" v2.6.0 is released I could have a version with Bluetooth and in German, thanks. Transmitter: TX16S

Done! I had heli on the brain so did a noheli build before I realised you wanted bluetooth=y,  not `heli=n` lol... so you can have both bluetooth with or without heli. 

[firmware_tx16s.zip](https://github.com/EdgeTX/edgetx/files/7930512/firmware_tx16s.zip)

JimB40, 2022-01-25:
> local charNames = {
>   "CHAR_RIGHT",
>   "CHAR_LEFT",
>   "CHAR_UP",
>   "CHAR_DOWN",
>   "CHAR_DELTA",
>   "CHAR_STICK",
>   "CHAR_POT",
>   "CHAR_SLIDER",
>   "CHAR_SWITCH",
>   "CHAR_TRIM",
>   "CHAR_INPUT",
>   "CHAR_FUNCTION",
>   "CHAR_CYC",
>   "CHAR_TRAINER",
>   "CHAR_CHANNEL",
>   "CHAR_TELEMETRY",
>   "CHAR_LUA"
> }

Hey Jesper,
Up to date I had it hardcoded with \xxx syntax
Testing it but once SMLSIZE flag is set those symbols do not appear on screen.
Haven't tested yet but I think setting any font size flag will do the same.

jfrickmann, 2022-01-25:
I am not sure that the char codes are the same on all transmitters, that’s why the constants are handy. 
And in any case, the special chars are only available for standard and BOLD fonts.

koaxheli, 2022-01-27:
This compilation would be ideal.
Thanks

cmake -Wno-dev -DPCB=X10 -DPCBREV=TX16S -DTRANSLATIONS=DE -DBLUETOOTH=Y -DLUA_MIXER=Y ..

pfeerick, 2022-01-29:
Done... here you go ;)

[firmware.zip](https://github.com/EdgeTX/edgetx/files/7962716/firmware.zip)

koaxheli, 2022-01-29:
Thank you very much

gagarinlg, 2022-09-15:
Is there any reason this issue is still open?

closed 2022-09-17 by gagarinlg