EdgeTX/edgetx #1318: Lua mixer scripts can't be deleted (mixer-script line popup menu)

wimalopaan, 2021-12-30:
1) the name `Reset` of the entry is misleading, it should read `Delete line`

2) the closure erases a local copy of the ScriptData because it captures that per value.

https://github.com/EdgeTX/edgetx/blob/d37b3e40518af42b6e54b3bda1b432a8f0431427/radio/src/gui/colorlcd/model_mixer_scripts.cpp#L259-L269

Fix using a pointer copy: 
```
   ScriptData* sp = &sd;
    button->setPressHandler([=]() -> uint8_t {
      Menu* menu = new Menu(window);
      menu->addLine(STR_EDIT, [=]() { editLine(window, idx); });

      if (runtimeData != nullptr) {
        menu->addLine(STR_RESET, [=]() {
          memset((void*)sp, 0, sizeof(sd));
          // TODO: anything else? reload scripts???
          storageDirty(EE_MODEL);
          rebuild(window, idx);
        });
        return 0;
      }
      return 0;
    });
```
But: I see this pattern all over the place (capture by value), which make me wonder what the reason is, maybe because there are different threads involved?

So, I don't know if a captue by ref would be ok?

eshifri, 2021-12-30:
Can you please provide description of the problem?
(Please notice, that your code does exactly the same as original: sp==&sd)

pfeerick, 2021-12-30:
There's some discussion on Discord in #dev. But basically, as mentioned the Reset context menu entry is really a delete function, and is currently ineffective.

raphaelcoeffic, 2021-12-30:
> There's some discussion on Discord in #dev. But basically, as mentioned the Reset context menu entry is really a delete function, and is currently ineffective.

Sounds like we should do something about it.

wimalopaan, 2021-12-30:
> Can you please provide description of the problem? (Please notice, that your code does exactly the same as original: sp==&sd)

The code I posted above does a different thing as the original. 
This the capture of the lambda is by-value, the original closure deletes a copy of the ScriptData, which effectively does nothing.
The modification passes the pointer (not the to be deleted object) by-value, and therefor really zeroes the array-element.

wimalopaan, 2022-01-01:
Is incorcoparetd in #1317 now

wimalopaan, 2022-01-03:
Maybe #1340  is of interest also.

philmoz, 2023-10-06:
Can this be closed?