EdgeTX/edgetx #1324: LVGL libopenui integration

gagarinlg, 2021-12-31:
libopenui and all widgets need to be ported to lvgl.

This issue needs some discussion on the software design.

- What is libopenui needed for in the future
- use lvgl screens. One screen, that is always there, for the main view, create a new screen and make that active for menus, warning screens, ...
- make that screen handling part of libopenui
- ...

Linjieqiang, 2022-01-07:
Why not only use the LVGL as the edgetx GUI, not include libopengi?

raphaelcoeffic, 2022-01-07:
> Why not only use the LVGL as the edgetx GUI, not include libopenui?

Because that amounts to a full re-write of the UI. Might happen in the future, but it would be good if we could avoid that.

closed 2022-06-11 by gagarinlg