EdgeTX/edgetx #1325: Yaapu screen not not populating

Flierman1945, 2021-12-31:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - EdgeTX version (e.g. 2.4.0 or nightly and #commit number)
 - Hardware (e.g. Frsky X9D+ or RM TX12S)
 
Selecting the Yaapu config, yaapu maps, or Yaapu debug, the variable values are visable when scrolled, but there is no left menu items.  Also the main yaapu sereen that displays maps bat volts home etc is no populating correctly

gagarinlg, 2023-02-26:
Is this an issue with 2.8.1?

Flierman1945, 2023-02-27:
I am on  Edge ver 2.8.0 and Yappu ver 1.3.0 and it all works ok.
With Edge ver 2.8.1 I have no info as I have not updated my test tx

pfeerick, 2023-02-27:
It should be no different for 2.8.1 (but please let us know if that's not the case!). Closing as obviously the situation has changed since 2.6.0 ;)

closed 2023-02-27 by pfeerick