EdgeTX/edgetx #1327: Error in building main under Ubuntu 20.04

nostromo123, 2021-12-31:
Fresh and updated Xubuntu 20.04 in vbox under Debian Bullseye host.

Building environment set up according to Instruction

https://github.com/EdgeTX/edgetx/wiki/Build-Instructions-under-Ubuntu-20.04

Errors show up when executing

$ make -j`nproc` firmware

Please could someone here have a look.

[Neue Datei.txt](https://github.com/EdgeTX/edgetx/files/7796476/Neue.Datei.txt)

raphaelcoeffic, 2021-12-31:
The important part is here:
```
clang.cindex.LibclangError: libclang-11.so: cannot open shared object file: No such file or directory. To provide a path to libclang use Config.set_library_path() or Config.set_library_file().
```

Did you update your system recently? Can you please check which versions of LLVM / clang are installed?

gagarinlg, 2021-12-31:
I had the same problem on Ubuntu 20.04 a few days ago. Installing libclang11-dev solved it. Some new build dependency must have ween intr2.

nostromo123, 2021-12-31:
The Xubuntu 20.04 system is absolutely fresh and updated.

Please see attached pic
[image]

nostromo123, 2021-12-31:
OK. I'll purge clang-10 and change to clang-11 and report back ....

raphaelcoeffic, 2021-12-31:
> OK. I'll purge clang-10 and change to clang-11 and report back ....

It's probably the python module which was updated.

nostromo123, 2021-12-31:
@gagarinlg

Your info did it. Thx!

The following packages were installed:

[image]

I am not up to find the source of that issue, but now the command 

$ make -j`nproc` firmware

worked! Hope, this is the only issue ;-)

closed 2022-01-02 by raphaelcoeffic