EdgeTX/edgetx #1330: Fill mixer-script-line %-value with useful information

wimalopaan, 2022-01-01:
What should be the purpose of the %-value at the end of each mixer-script line? 

It displays the member `ScriptInternalData.instructions`. This maybe a usefull information to get an overview how computational intense a script can be, but this field is never  gets filled with something. 
In `luaTask()` a measure is calulated but I can't find a way to reference the script data structure.

wimalopaan, 2022-01-01:
@jfrickmann Do you have any hints here?
If not, then I suggest to remove this value at all for the moment.

jfrickmann, 2022-01-01:
No, I suggest that we remove it

pfeerick, 2022-01-02:
I agree, it should go if there is nothing useful to fill that space.

wimalopaan, 2022-01-03:
Fixed in #1317

wimalopaan, 2022-01-03:
Maybe #1340  is of interest also.

closed 2022-03-23 by raphaelcoeffic