EdgeTX/edgetx #1333: Potential out of bounds access of flightmode name on main screen

wucke13, 2022-01-01:
*Before submitting,  search the open issue if there is already a report of the bug*

**Describe the bug**
I do happen to have two flightmodes with rather long names, `PID-Adjust` and `HeadAdjust`. Now, on the main screen, bottom, center normally the flightmode is displayed. However, both Flightmodes have an extra character at the end: `HeadAdjust)` &  `PID-Adjust*`. In the model page for the flightmodes everything is perfectly fine. Now my suspicion tells me somewhere in the code a char array is not null terminated (to save space?) and somewhere else this char array is assumed to be null terminated, thus resulting in a out of bounds access. Or is there some buffer offset just wrongly defined, a classical of-by-one?

**To Reproduce**
Create  a flight mode with the maximum number of characters in it's name (10?). Observe weird characters being appended.

**Expected behavior**
Only the characters which belong to the flightmode name are shown

**General information**
 - EdgeTX version 2.5.0 28cdb40e
 - Hardware FrSky X10S

wimalopaan, 2022-01-01:
I think we have the same fixed-string mess here. Will check for that ...

raphaelcoeffic, 2022-01-01:
> I think we have the same fixed-string mess here. Will check for that ...

This trick is used everywhere. The issue is in the code using that string, not the string itself: it is as designed.

wucke13, 2022-01-02:
What is meant with "this trick"? Just in case that it refers to char arrays as strings, which are not null terminated: I'm totally in favor of it, e.g. having as few potentially unbounded strings (only null terminated) as possible.

pfeerick, 2022-01-02:
Yes, that what it is in reference to. There are a lot of strings that are deliberately not null terminated in order to use as little memory as possible.

wimalopaan, 2022-01-02:
Should be fixed in #1337

raphaelcoeffic, 2022-01-02:
> What is meant with "this trick"? Just in case that it refers to char arrays as strings, which are not null terminated: I'm totally in favor of it, e.g. having as few potentially unbounded strings (only null terminated) as possible.

Using `char` arrays that are either 0-terminated, or full (using the full fixed-size space without 0-termination). This means that every time you copy these strings elsewhere, you need to use something like `strncpy(dest, src, sizeof(src))`, assuming `src` is not just a pointer, but something the compiler can actually determine the size of (`char` array). If the last assumption does not hold, you need to use one of the `#define` to force the maximum size. If `dest` needs itself to be 0-terminated, something like `dest[sizeof(src)] = '\0';` should be used afterwards.

wimalopaan, 2022-01-02:
There are many pitfalls with this, as we see often such errors not correctly handling these kinds of "strings". Personally I find this a hack. But the source is as it is now ;-) 

Clearly it would be better here to use things like `std::string_view` or `FixedString<N>` with the max-length encoded into the type. With modern C++ (since C++17) most of the fiddling could be done safe at compiletime (as a replace to `snprintf()`, et.al.).

gagarinlg, 2022-01-02:
It is a hack if you have lots of memory available. In embedded syswtems with memory restrictions this is used all the time, because very byte counts. And when you take alook at the history of EdgeTX/OpenTX you will find a lot of devies with very little ram.

wimalopaan, 2022-01-02:
> It is a hack if you have lots of memory available. In embedded syswtems with memory restrictions this is used all the time, because very byte counts.

Sure, I know that.
As I said: with "modern" C++ (>= C++17) one could avoid that. And yes, this is not possible for a code-base like these, because is is an enormous effort and the project actual sticks to C++11.

wimalopaan, 2022-01-02:
I saw may situations where:

```
str::string{text, MAX_TEXT_LEN};
```

is used and that is (mostly) a waste of memory, because in this case all chars (including potential '\0') are copied intoi the string.

Something like this would help:

```
template<size_t N>
constexpr std::string stringFromNtString(const char (&a)[N]) {
    return std::string(a, strnlen(a, N));        
}    
```

pfeerick, 2022-01-02:
Unfortunately, C++17 didn't exist when all of this was written. 😢  Hopefully someday we can transition to that has nice things like this

wucke13, 2022-01-02:
Out of curiosity,

> the project actual sticks to C++11

Why?

I come from a Rust background, and also have a modest idea of modern (post C++ '17). Whatever safety can be gained (at compile time even without performance penalty) seems worth it, so why would the project stick to C++ '11 despite some very nice advancements towards compile-time safety checks in more recent C++ versions?

raphaelcoeffic, 2022-01-02:
> Out of curiosity,
>
>
> Why?
>
> I come from a Rust background, and also have a modest idea of modern (post C++ '17). Whatever safety can be gained (at compile time even without performance penalty) seems [...]

One thing at a time ;-) The transition to C++11 is not even digested yet. But no worry, it's coming.

wucke13, 2022-01-03:
> I saw may situations where:
> 
> ```
> str::string{text, MAX_TEXT_LEN};
> ```
> 
> is used and that is (mostly) a waste of memory, because in this case all chars (including potential '\0') are copied intoi the string.
> 
> Something like this would help (a little bit):
> 
> ```
> template<size_t N>
> constexpr std::string stringFromNtString(const char (&a)[N]) {
>     return std::string(a, strnlen(a, N));        
> }    
> ```

Just a comment here: from a safety/reliability standpoint, when using heap memory this wasteful usage might not be so bad. In order to gain confidence one should try to make the application consistently show its worst performance re memory usage. If it then still does not OOM, one can be reasonably sure that it under normal circumstances never will. Of course that's a rough approximation and not necessarily the right course for EdgeTX, but reproducible doing worst case allocations is a thing.

wimalopaan, 2022-01-03:
Maybe #1340  is of interest also.

closed 2022-01-09 by pfeerick