EdgeTX/edgetx #1334: Mode 2 and Mode 4 options are swapped over

Aaronsss, 2022-01-01:
**Describe the bug**
Mode 2 and Mode 4 options seem to be the wrong way around. I tried playing about with the default channel order on the radio but it didn't seem to change anything

Mode 2 should be throttle and yaw on the left stick, pitch and roll on the right stick

**General information**
 - EdgeTX version (e.g. 2.4.0 or nightly and #commit number) 2.6.0 RC1
 - Hardware (e.g. Frsky X9D+ or RM TX12S) NV14

eshifri, 2022-01-01:
Can you please provide some details/describe the sequence?
I cannot reproduce it: for mode 2 throttle is channel 3 and is on the left stick; rudder is channel 4 is on the left stick as well.

Aaronsss, 2022-01-01:
With the left stick pushed to top right:
With mode 2 selected on the radio:
[image]
With mode 4 selected on the radio:
[image]

The transmitter has been set to AETR as well.

eshifri, 2022-01-01:
Thank you!
Can you please take a look at the channel monitor on the thee TX as well?

Aaronsss, 2022-01-01:
Mode 2
[image]

Mode 4
[image]

eshifri, 2022-01-01:
Looks convincing, but I still cannot reproduce it. :-)

rotorman, 2022-01-01:
@Aaronsss what EdgeTX FW version you are running? What does the Info tab say (GitHub commit hash)?

Aaronsss, 2022-01-01:
[image]

rotorman, 2022-01-01:
Thank you for the pic above! How is the mapping on the analogs diagnose page? Which channel corresponds to which stick and axis?

Aaronsss, 2022-01-01:
On the analogs test page:
01 = yaw
02 = throttle
03 = pitch
04 = roll
05 = VRA
06 = VBR

rotorman, 2022-01-01:
The order of sticks matches my FlySky gimbal order, thus I wonder if you just have some settings wrong.
Can you try the following - take the SD card of your NV and rename \MODELS to \MODELS_old and \RADIO to \RADIO_old. Put the card back in radio and re-run the calib, set your more and stick order and create a test model. Do the channels match now?

Aaronsss, 2022-01-01:
Ok I have figured out what has happened. When I flashed the radio the default channel map was RETA. I updated to AETR but from this point on I was copying the model01. 

Is there a way to update the channel map of each model that has already been created? It has taken me the best part of 2 days to get my models back to almost where they were on the old flysky version so don't want to do that again.

rotorman, 2022-01-01:
The radio and model settings of 2.6.0rc1 are in YAML - a human readable file that you can edit with a text editor (just make a backup first, in case you mess something up, you can always go back to old state).

Aaronsss, 2022-01-01:
Ah I have figured it out I just need to swap RUD and AIL on the inputs tab so that fixes the issue for me.

Is it worth altering the code so that AETR is the default upon first flash to prevent others having the same problem?

raphaelcoeffic, 2022-01-01:
> Ah I have figured it out I just need to swap RUD and AIL on the inputs tab so that fixes the issue for me.
> 
> Is it worth altering the code so that AETR is the default upon first flash to prevent others having the same problem?

The order you enter applies only to new models, not in general, as you can freely map whatever you want to the channels you want.

what would make sense however would be to ask the user for that setting before creating the first model.

pfeerick, 2022-01-01:
Indeed... and if this isn't an issue already was something I mentioned early on... I'll work on this... as I want to sort out the calibration page anyway... has one or two things I really find annoying...

closed 2022-01-02 by pfeerick