EdgeTX/edgetx #1335: Screen doesn't respond if you try to shut the radio down with an armed quad connected (NV14)

Aaronsss, 2022-01-01:
**Describe the bug**
If the screen has turned off on the NV14 and you try to turn the radio off with the quad armed when the confirmation message comes up you can't click the button to turn the radio off

This problem does not exsist if the screen was on when you shutdown the radio

**To Reproduce**
Let the screen turn off with it set to only come back on if you touch the screen, arm the quad (props off ofc) turn off the radio. The shutdown screen appears followed by the confirmation screen. When you try to press the ok button you will be unable to do so

**Expected behavior**
The radio should turn off when the button is pressed on the confirmation message. Perhaps the touch screen part has been disabled.

As a side note I feel like the confirmation message should have a shutdown and don't shutdown button incase you want to keep the radio active

**General information**
 - EdgeTX version (e.g. 2.4.0 or nightly and #commit number) 2.6.0 RC1
 - Hardware (e.g. Frsky X9D+ or RM TX12S) NV14

pfeerick, 2022-01-04:
Ok, I can repeat this, and I suspect the TX16S will suffer from it as well. As you point out, it seems like a specific order of steps is needed - you need to attempt power off when the backlight is off, at which point the shutdown animation will be shown, and then as soon as that completes, the backlight will go off again. if you touch the screen or move the sticks the backlight will come on, but you are unable to tap the arrow to confirm power off. 

You can use the left and right trim hat buttons, but that also wont be an option on future touch-only radios, so the yes / no type conformation on screen is indeed needed, and is probably better to have generally

pfeerick, 2022-03-19:
I can't pin down the exact commit, but I believe this has now been resolved, as I'm no longer able to reproduce this - on either the NV14 or TX16. If memory serves me right, there was a commit that ensured the backlight would stay on while alerts like this were shown, and that was responsible for curing this, as it prevent touch from being disabled. 

Anyway, please don't hesitate to comment or reopen this issue if that is no the case, but then we'll have to tighten down on the exact sequence needed to reproduce this.

closed 2022-03-19 by pfeerick