EdgeTX/edgetx #1336: santa 2.6

bjou12, 2022-01-01:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - EdgeTX version (e.g. 2.6=.0 or nightly and #commit number)
 - Hardware (e.g. radiomaster+ or RM TX12S)
 quad feel not locked in lots of latency

pfeerick, 2022-01-01:
Need a little more information before anything can be done about that. Out of the questions from the template you've said what version of the firmware you're running. What transmitter is this on? What RF system (protocol/tx module and receiver)? What version of betaflight? Is this new behaviour? If so, what did you change?

gagarinlg, 2023-02-26:
closed, no response

closed 2023-02-26 by gagarinlg