EdgeTX/edgetx #1343: ERLS configurator laggy / EXT module serial packet loss

palladin82, 2022-01-04:
Hardware RM TX16S
Telemetry laggy, ERLS2.lua laggy. As a result, the hardware constantly freezes in scripts and widgets.
Tried 2.4 - all is good!
In nightly#a077f2e & 2.5 tried to change the speed of the serial port of the external module, there is no result.
Please pay attention.

pfeerick, 2022-01-04:
I have been running 2.5 with ES24TX ELRS regularly on TX16S and noticed no lag issues with telemetry or ELRS Lua. Was running v1.2.0 and v2.0.0 since release. I have not upgraded to v2.0.1 yet, and have seen a number of reports of issues there which may be hardware specific or configuration issues, so am holding off for now.

palladin82, 2022-01-05:
I'll try to downgrade 2.0.0, and inform additionally.

palladin82, 2022-01-05:
Yes, there is a problem in ELRS, [link on ELRS](https://github.com/ExpressLRS/ExpressLRS/issues/1248)

pfeerick, 2022-01-05:
Ok, hopefully the downgrade alleviates it for you, and the devs can then figure out what is going on between ETX and ELRS that is causing this.

palladin82, 2022-01-05:
No luck.
[2.6 rc2](https://youtu.be/EOrpPxDPX7U)
[2.4.0](https://youtu.be/pbm-345XCms)
same config...

palladin82, 2022-01-07:
Found the reason why the configurator was lagging.
I'm using Yaapu telemetry widget, and it was active on one of the screen. When I disable it, configurator works well.

pfeerick, 2022-01-07:
Ah, that makes sense... Yappu will be trying to grab the telemetry data, and then ELRS configurator goes and re-configures stuff when it starts and throws the whole joint out of wack.

palladin82, 2022-01-08:
I remember some time ago the widgets went into disabled mode when the TX was in setup mode. Isn't it?

pfeerick, 2022-01-08:
Perhaps if you were configuring the widgets (i.e. at the screen setup page), but I don't know the rest of the time. I'm not sure it would be practical either as that means widgets that report telemetry (e.g. yappu) would stop working as soon as you entered the setup page - instead of continue to run and report values.

gagarinlg, 2023-02-26:
This is fixed with 2.8

closed 2023-02-26 by gagarinlg