EdgeTX/edgetx #1346: INAV Telemetry LUA script stopped working between 2.6-RC1 and 2.6-RC2

JoeriCapens, 2022-01-04:
**Describe the bug**
After upgrading to RC2, the widget now shows: ERROR in create(): ?:0: attempt to compare nil with number

**To Reproduce**
Flash 2.6-RC1, set up the INAV widget (https://github.com/iNavFlight/OpenTX-Telemetry-Widget) on a full-screen secondary layout, this works. Flash the 2.6-RC2 update, it now shows an error.

**Expected behavior**
The INAV Telemetry LUA script should still work in EdgeTX 2.6-RC2

**General information**
 - EdgeTX version: 2.6-RC2
 - Hardware: Radiomaster TX16S

pfeerick, 2022-01-04:
Thanks to some [investigation by stronnag](https://discord.com/channels/839849772864503828/842693696629899274/927954865643196458), it looks like recent changes to getFieldInfo are the cause :( 

> so it seems that getFieldInfo does not always return a unit field, which rather breaks opentx scripts that assumes it does (as it did previously).

Meaning #1300 is to blame

RipperGOT, 2022-01-05:
I have tried it and the problem is reproduced.

stronnag, 2022-01-05:
There is a workaround (fix), including a new Zip of the inav widget in the inav repo.
Just checks the field exists before trying ti use it.

https://github.com/iNavFlight/OpenTX-Telemetry-Widget/pull/24

jfrickmann, 2022-01-05:
I will try to look into it if I can, but right now, I have a few problems. For several months, the little simulator windows like the telemetry simulator have been completely frozen on my builds under Linux in Win11/WSL. I don't know why, and I do not know how to fix that. And without that working, I cannot troubleshoot iNav. I can wait for someone to provide a Windows build, but then I will not be able to add traces and test fixes, as I cannot build directly under Windows.
So I am sorry to say, but this could take a while...

stronnag, 2022-01-06:
As far the the inav lua widget mainatiners are concerned, this is a non-issue. We understand the problem and have a trivial workaround that maintains backwards compatibility with OTX and older ETX.

jfrickmann, 2022-01-08:
Thanks for letting me know that! Although it is not currently urgent, I will still try to find out what is going on before others get the same issue

jfrickmann, 2022-01-15:
OK, I have managed to look into the issue now, and I think that it is a feature, not a bug 🤣

The main purpose of #1300 was to make it possible to also allow source IDs as input for `getFieldInfo`. And it works very well, because when you have a sensor named "0007", then it looks up `getFieldInfo(7)` and returns the sensor `input7`. And iNav/data.lua barfs on line 12...

So the question is now - shall I try to convince Lua not to convert "0007" to 7, or shall we accept it as a feature and update our Lua scripts accordingly?

stronnag, 2022-01-15:
The problem for the inav script was that the documented/ legacy  behaviour changed. 

The [documented behaviour](https://luadoc.edgetx.org/part_iii_-_opentx_lua_api_reference/general-functions-less-than-greater-than-luadoc-begin-general/getfieldinfo) is that `getFieldInfo` either returns `nil` or it returns a table with a `unit` key; it this instance it returns a table without the `unit` field. This caused the inav script to fail.

If breaking legacy APIs is now a 'feature', then so be it. I have no view on how you should proceed; however you may wish to consider that 3rd party script authors may not always share your enthusiasm to "update our Lua scripts accordingly".

jfrickmann, 2022-01-15:
`getFieldInfo` has never returned a table with `unit` for non-telemetry sources. Here is the code for OpenTX:
https://github.com/opentx/opentx/blob/8088e78c980dc6ef65203434d5aa7d3aba10ad54/radio/src/lua/api_general.cpp#L909-L915
It is somewhat confusing that they have made a function `lua_pushtablenil`, because setting the value of a table field to `nil` will delete the field. And adding a field with a `nil` value will have no effect whatsoever.
So what I am saying here, is that most likely, you are trying to look up e.g. the field "0007", and instead you get the non-telemetry field with ID=7, which appears to be `input7`. And since this is not a telemetry field, you do not get a `unit` field.
And moreover, your fix does not work, because you are looking at the wrong field to begin with.
I will take a look at the code and see if I can make Lua a little less "helpful", so it does not convert names that could be numbers to numbers.

stronnag, 2022-01-15:
Regardless, this is a regression. The extant code has worked for more than 4.5 years, as documented. Even in ETX < 2.6rc2. Now it doesn't.

jfrickmann, 2022-01-15:
Yes, and I just proposed a fix for that, which I am about to start working on.

jfrickmann, 2022-01-15:
OK, iNav people - please check #1423, and let me know if that solved the problem.

RipperGOT, 2022-01-16:
Not only in Inav widget, BF conf #1427
It's the same?

closed 2022-01-17 by pfeerick